PX4/PX4-Autopilot #1565: Check that the RTC time is set correctly by GPS

LorenzMeier, 2015-01-03:
Current master has RTC support. It will retain the wall clock for a few days on FMUv2 hardware. Without setting it initially, it currently reads:

```
nsh> date
Jan 01 00:08:27 2000
```

The GPS driver has support for setting the time. It would be interesting to know if the date output changes.

LorenzMeier, 2015-01-03:
@sjwilks In case you can get lock with a setup that has a terminal connected it might be a quick test.

bansiesta, 2015-01-04:
This works for me, even after a power cycle.

LorenzMeier, 2015-01-04:
So you're getting a 2015 date, not 2000, correct?

LorenzMeier, 2015-01-04:
With current master - I fixed series 6-7 receivers just half an hour ago.

mhkabir, 2015-01-04:
I tested with my timesync system, and I confirm that the date is correctly set by the companion computer. Uses similar method of setting as GPS.

LorenzMeier, 2015-01-04:
Also works here:
```
nsh> date
Jan 04 11:08:57 2015
```

closed 2015-01-04 by LorenzMeier