PX4/PX4-Autopilot #1571: Set up RTC / time correctly in presence of GPS AND companion computer

LorenzMeier, 2015-01-04:
@mhkabir This one seems naturally for you. Currently your time sync code and the GPS driver both call the system time interface. We would need to work out who has precedence.

Note that the GPS time runs off the GPS time pulse which in turn runs off atomic clocks in space. So the odds that the companion computer time is more accurate goes to zero. In fact, we actually would want to sync it its time based on GPS time if possible.

mhkabir, 2015-01-04:
@LorenzMeier The GPS clock gets precedence. I made sure that happens. The companion clock is only used one time when the first time message arrives, to get the clock in PX4 correct for logs, etc. . It is not augmented further. If GPS is present, GPS driver keeps correcting.

That reminds me, I need to PR to sdlog2 for it to use the time from RTC, and not directly from GPS.

mhkabir, 2015-01-04:
Also, the companion clock is skewed to conform to GPS time if a certain ROS node is running.

LorenzMeier, 2015-01-04:
Wohoo! Awesome. Closing.

closed 2015-01-04 by LorenzMeier