PX4/PX4-Autopilot #1583: Store sensor ID during calibration, fail preflight check if ID changes

LorenzMeier, 2015-01-04:
This is to address #1045. The device ID is already filled correctly, so this one should be fairly quick / easy.

@bansiesta Interested? Anyone else interested? Its nice self-contained.

LorenzMeier, 2015-01-06:
Initial code hint:
https://github.com/PX4/Firmware/blob/master/src/drivers/device/device.cpp#L100-L104

hauptmech, 2015-01-07:
I'm looking at this but not sure if I have the systems knowledge yet. If someone else can beat me to it, please do.

hauptmech, 2015-01-07:
When redundant mags are used one would need a calibration for each sensor.

What is being stored is not a sensor ID but rather sensor bus and type. It is possible for field hardware changes (swap 3DR GPS for instance) to allow the preflight to pass. 

Unfortunately I think few of the sensor have true unique id's. An interesting approach might be to collect signatures of the sensors (essentially an online collection of the calibration data) and warn when the signature goes outside the bounds of the expected calibration.

LorenzMeier, 2015-02-10:
Thanks for helping out there! All implemented now.

closed 2015-02-10 by LorenzMeier