PX4/PX4-Autopilot #1585: Remote Euler angles from logfiles

LorenzMeier, 2015-01-05:
@tumbili @thomasgubler we discussed this a while back, did you proceed with this? As it will induce confusion during the transitional phase, it would be good to digest this rather sooner than later.

@DrTon does FlightPlot support quaternion conversion already?

tumbili, 2015-01-05:
I started this indeed. You said that you would like a vtol status topic publishing a quaternion of the transformation. Do you mean the quaternion of the transformed attitude? Because the transformation cannot really be expressed as a rotation it is really more a change of body frame axes. We could instead introduce three new parameters in the vtol module which indicate the change of axes from one mode to the other, e.g.
[1,2,3] -> no transformation, same attitude
[-3,2,1] -> x axis becomes negative of old z axis, y stays the same, z becomes old x axis

DrTon, 2015-01-05:
>@DrTon does FlightPlot support quaternion conversion already?

Just implemented it in FlightPlot (in master but not yet in binary), only need to adjust default fields names and test.

>@tumbili, why not to use "fixed wing style" attitude and just add rotation in attitude and position controllers for VTOL mode? In this case attitude and attitude setpoints will be consistent and you will always clearly determine orientation of vehicle. To avoid singularities we can use quaternions, and to display orientation of VTOL in "copter style" I can add additional rotation to FlightPlot.

Using two attitude representations for the same vehicle is very confusing IMO...

tumbili, 2015-01-05:
@DrTon This generally sounds like a good idea keeping in mind the following:
The reason why we transformed attitude was because it allowed us to use the conventional body axes of both multicopter and fixed wing. Therefore, we only had to make minor changes in the fw_att_contoller to make things work. If you want to use one set of body axes then you need to transform all quantities for either fixed wing or multicopter module. If this is not related with unnecessary complexity then I agree to go with your proposal.

LorenzMeier, 2015-01-05:
There is no complexity associated with it. The VTOL controller should emit the VTOL status topic, which allows the fixed wing controller to pick the transformation off it and apply it once. Same for logging: VTOL status holds the transformation and the plotter applies it locally.
The current state was ok to get things up and running, but it ties layers and apps together that have nothing to do with each other, hinting that we need to improve its architecture.

Since we apply axis changes to sensors based on a rotation matrix, I'm not sure why we couldn't do the same for the VTOL, if its done in the right coordinate frame.

tumbili, 2015-01-05:
@LorenzMeier Basically what your are saying is the same that we have at the moment:
Fixed wing attitude controller checks if is_vtol= true and if yes it applies the transformation (hardcoded though). Your proposal is just a bit cleaner but I wouldn't say it breaks any ties between layers.
And with rotation you probably mean something like this https://github.com/PX4/Firmware/blob/master/src/lib/conversion/rotation.cpp which is exactly what is done in the fw_att_control module at the moment just only for one case.

@DrTon @LorenzMeier I think in the end we will be looking (at least I hope so) for a controller that can handle a VTOL as ONE vehicle without always having to differentiate between fixed wing and VTOL.

closed 2016-05-22 by LorenzMeier