PX4/PX4-Autopilot #1597: Mixer file names start with FMU but actually provide mixers for px4io

anton-matosov, 2015-01-06:
Currently I am working on the project that requires almost all the PWM outputs, thus I need to provide mixers for all of them. So I am going to update current init scripts to provide capability for configuring pwm outputs of both px4io and px4fmu devices.
While doing this I have noticed that existing mixer file names start with FMU but actually provide mixers for px4io. This brings in a confusion and makes it hard to give a name for real FMU outputs mixer file.

I have a proposal (and will contribute this change) to remove FMU_ prefix from the existing mixer files and introduce the double extensions to distinguish between fmu and io mixers, e.g.:
FMU_AERT.mix -> AERT.io.mix
and if it needs extra outputs: 
AERT.fmu.mix 
can be provided

LorenzMeier, 2015-01-06:
Have you seen the FireFly PR? It does something equivalent. I will rebase that one, and after you can base off and do the proposed cleanup.

On the file naming, I would prefer if mixers were only amended with an additional name part of there are two parts for the vehicle - maybe this is what you meant.

LorenzMeier, 2015-01-06:
This is the branch I suggest to base on. If we have positive testing feedback I'll merge it.

LorenzMeier, 2015-01-06:
https://github.com/PX4/Firmware/pull/1598

anton-matosov, 2015-01-06:
That's the great news!! @LorenzMeier  and @tumbili thanks guys, you implemented exactly what I needed )))

---

Regarding the naming:
I have initially proposed the following naming convention:
quad_x.io.mix     (1)
quad_x.fmu.mix

But it does involve internal details (fmu, io) instead of external px4 outputs naming (main, aux). So another convention can be used:
quad_x.main.mix     (2)
quad_x.aux.mix

or with name part instead of double extension:
quad_x_main.mix     (3)
quad_x_aux.mix

or we can omit the _main part:
quad_x.mix     (4)
quad_x_aux.mix

---
I would prefer the (2) option.
What would be your choice?

LorenzMeier, 2015-01-06:
Hi,

2) sounds fair, however, each mixer file adds to the binary size. So the aux files should only by created where needed (e.g. for the FireFly or for your vehicle).

anton-matosov, 2015-01-06:
Sure, there will be no empty files, the only thing I will do is change the naming.

closed 2015-01-10 by anton-matosov