PX4/PX4-Autopilot #1609: Adjust handling of L1 radius

LorenzMeier, 2015-01-07:
Right now the navigation logic is not consistent. Need to adjust L1 radius / switch point. This is what we initially had and what is required to guarantee reachability.

thomasgubler, 2015-01-07:
We still have the functions to calculate this value min(wp_radius, l1_distance) in the l1 controller. It's just a matter of feeding the navigation_capabilities topic into the navigator mission_block.

thomasgubler, 2015-01-07:
The value is calculated and published here: https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L1474

I suggest to publish the l1_distance as turn distance and then calculate min(wp_radius, turn_distance) in the navigator where the wp_radius is anyway available.

LorenzMeier, 2015-01-07:
@thomasgubler Thanks!

LorenzMeier, 2015-07-03:
Implemented and fixed, closing.

closed 2015-07-03 by LorenzMeier