PX4/PX4-Autopilot #1610: Servo gimbal controller

anton-matosov, 2015-01-07:
Hi everyone,

I am starting developing servo gimbal controller.
I plan to adopt code from the AMP project:
https://github.com/diydrones/ardupilot/tree/master/libraries/AP_Mount
or at least use it as a reference.

You are welcome to post any objections or proposals into this ticket.

LorenzMeier, 2015-01-08:
Sounds great! We currently can't accept GPL code into the codebase, as its incompatible with BSD. So you would need to write it yourself.

Note that I wrote a servo gimbal controller a year ago. Give me a day to dig it out and put it onto a branch.

anton-matosov, 2015-01-08:
Ok. I missed the incompatible licenses. Thanks for pointing me on that.

I will wait for your code of gimbal controller to start with.

anton-matosov, 2015-01-09:
Hi,
Did you get a chance to find the code of your gimbal controller?

anton-matosov, 2015-01-10:
@LorenzMeier do you have any updates?

LorenzMeier, 2015-01-10:
Yep, sorry about that. Will do this tomorrow morning as soon as possible.

anton-matosov, 2015-01-10:
Ok, thanks.

anton-matosov, 2015-01-11:
@LorenzMeier do you have any updates? =)

LorenzMeier, 2015-01-11:
Sorry about the latency on this!
Here it is:
https://github.com/PX4/Firmware/pull/1643

Many months have passed since I last used it, but it did work fine back then.

anton-matosov, 2015-01-11:
Thanks. Really good starting point for me =)

LorenzMeier, 2015-06-14:
@anton-matosov What's your current status on this? Did you continue with it?

anton-matosov, 2015-06-15:
Hi, not really. I have no spare time for this project =(

LorenzMeier, 2016-03-24:
We have this in.

closed 2016-03-24 by LorenzMeier