PX4/PX4-Autopilot #1611: FW takeoff WP enforces 3D acceptance radius instead of altitude-only check

kd0aij, 2015-01-07:
MissionBlock::is_mission_item_reached() handles MC and FW takeoff WPs differently; for MC only altitude must be with acceptance radius, for FW this requirement is 3D. 

This would explain the unexpected circling I observed around a takeoff WP. Since QGC displays no radius for Takeoff WPs the implication is that there is no minimum radius.

Is this difference in behavior intentional?

thomasgubler, 2015-01-08:
I am not sure why we only have an altitude limit for MC. The idea is that you always takeoff vertically with MC. But this would not speak against a 3D limit itself. As you noted there is no value for the acceptance radius in the mavlink takeoff waypoint item. That's why NAV_ACC_RAD is used here. NAV_ACC_RAD is actually set to higher value in rc.fw_defaults if you configure your system. I suggest that we integrate the ideas from #1609 also in the takeoff waypoint handling (for FW).

LorenzMeier, 2015-01-08:
I think we should change the autoconfig logic. If you set the autostart ID manually and forget to set the autoconfig param, you can end up with a stale config. We could store the requested and set autostart IDs as parameters instead and apply autoconfig on change. That would enforce the setup as expected.

LorenzMeier, 2015-02-10:
I'm closing this as we would want to circle upwards if the takeoff alt WP is high. We did improve the auto start handling stuff a bit lately.

closed 2015-02-10 by LorenzMeier