PX4/PX4-Autopilot #1619: #If X==Y is true when both are not defined  use of -Wundef

davids5, 2015-01-08:
We ran into an issues with some of PX4 code where the construct !If X==Y is true when both are not defined.

For example:

//#define PIGS 1
//#define DOGS 2

#if (DOGS == PIGS)
#warning("DOGS == PIGS") // it is to a compiler
#else
#warning "DOGS != PIGS"
#endif
When then built with -Wundef to ferret out any others:

he consolidated results of the -Wundef:

warning: "__APPLE__" is not defined [-Wundef] APPLE
warning: "__EXCEPTIONS" is not defined [-Wundef] __EXCEPTIONS
warning: "__linux" is not defined [-Wundef] __linux
warning: "__linux__" is not defined [-Wundef] linux
warning: "_HAS_EXCEPTIONS" is not defined [-Wundef] _HAS_EXCEPTIONS
warning: "_WIN32" is not defined [-Wundef] _WIN32
warning: "_WIN64" is not defined [-Wundef] _WIN64
warning: "CONFIG_ARCH_BOARD_AEROCORE" is not defined [-Wundef] CONFIG_ARCH_BOARD_AEROCORE
warning: "CONFIG_ARCH_BOARD_PX4FMU_V2" is not defined [-Wundef] CONFIG_ARCH_BOARD_PX4FMU_V2
warning: "CONFIG_NSOCKET_DESCRIPTORS" is not defined [-Wundef] CONFIG_NSOCKET_DESCRIPTORS
warning: "CONFIG_STM32_TIM1" is not defined [-Wundef] CONFIG_STM32_TIM1
warning: "CONFIG_STM32_TIM8" is not defined [-Wundef] CONFIG_STM32_TIM8
warning: "MULT_COUNT" is not defined [-Wundef] MULT_COUNT
warning: "STM32_NGTIMNDMA" is not defined [-Wundef] STM32_NGTIMNDMA
warning: "UAVCAN_DEBUG" is not defined [-Wundef] UAVCAN_DEBUG
warning: "UAVCAN_MEM_POOL_BLOCK_SIZE" is not defined [-Wundef] UAVCAN_MEM_POOL_BLOCK_SIZE
warning: "UAVCAN_STM32_CHIBIOS" is not defined [-Wundef] UAVCAN_STM32_CHIBIOS
warning: "UAVCAN_STM32_TIMER_NUMBER" is not defined [-Wundef] UAVCAN_STM32_TIMER_NUMBER

Need to 1) see if any of them are significant 
2) see if the warnings can be remedied with an alternate construct of #if defined(XYZ) #if XYZ==PQT .... #endif #endif or such.

dagar, 2016-11-08:
Still an issue?

davids5, 2016-11-08:
It may be,

The best way to know is build with -Wundef to ferret out any any remaining.  

The fix is simple enough see if the warnings can be remedied with an alternate construct of 
#if defined(XYZ) #if XYZ==PQT .... #endif #endif or such.

dagar, 2017-02-22:
We should try -Wundef again to see the specific usage.

closed 2017-08-20 by LorenzMeier