PX4/PX4-Autopilot #1621: Upload latest master, beta and stable builds to AWS

LorenzMeier, 2015-01-09:
@dagar The encrypted AWS credentials are already present in Travis:
https://github.com/PX4/Firmware/blob/master/.travis.yml#L31-L39

I have given you access to use them in pushes to upstream branches (they are not available in PRs). Would you be interested to help set up the s3cmd (http://s3tools.org/s3cmd) to upload the binaries to AWS? I would like to recreate this:
http://px4.oznet.ch

Except flow, which obviously should be in its own Travis instance and can be ignored for now.

The goal of this is to have all building in one place.

dagar, 2015-01-10:
How many builds do you want to keep? Under a certain total size or drop them after a week, month, etc?

LorenzMeier, 2015-01-10:
Actually, given that Travis stores each build already in AWS (see the link at the bottom of the page) just keeping the latest might be enough.

dagar, 2015-01-14:
@LorenzMeier - almost working a22ea3244cf1a4d158456ad908d778bf80cf7764

Running into some problems with getting s3 to list all files in a directory. I tried using this https://github.com/rgrp/s3-bucket-listing, but haven't gotten it to work. Could you grant everyone the List and View permissions as described in that project?

LorenzMeier, 2015-01-14:
@dagar I have enabled listing. Let me know if that helped.

dagar, 2015-01-15:
Working, but still no directory listing. - http://px4-travis.s3.amazonaws.com/index.html

I was able to get a directory index working for my test s3 bucket by setting "Do not enable website hosting" under "Static Website Hosting" and then under "Permissions" make sure you have "List" and "View Permissions" set for "Everyone"

Once s3 works this - http://px4-travis.s3.amazonaws.com/s3deploy/index.html
should look more like this - http://px4test.s3.amazonaws.com/list.html

After that it can be limited to just stable/beta/master.

LorenzMeier, 2015-01-15:
Thanks, I assumed you actually wanted hosting. Here is the new output:
https://px4-travis.s3.amazonaws.com/index.html

dagar, 2015-01-15:
Yes, I assumed that as well. Anyway how is that? We'd need to let it populate the real branches, but it should be equivalent to http://px4.oznet.ch at this point.

LorenzMeier, 2015-01-15:
Cool! I just enabled Travis for Flow, does that help you to cut over the last bit to Travis as well? I have encrypted the AWS credentials the same way there, so you should be able to just copy & paste stuff from Firmware.
https://travis-ci.org/PX4/Flow

AndreasAntener, 2015-01-15:
Hi, I just wanted to let you know that I did something similar yesterday for FlightPlot but with a different build server (Jenkins). Although I didn't bother with the dynamic listing. The build server generates the "index.html" with new links after each build and uploads it so S3 along with the binaries. The wiki then includes this file. (See: https://s3-eu-west-1.amazonaws.com/flightplot.antener/releases/latest.html, https://pixhawk.org/dev/flightplot#download and http://jenkins.antener.name/). From the commit it looks like you'll do the same, this is what my build server does, maybe it's of use:

```
#!/bin/sh

# get tag name
TAG=`git describe --tags --always`
JAR_NAME="flightplot-$TAG.jar.zip"
APP_NAME="flightplot-$TAG.app.zip"

cd out/production

# rename artifacts
echo "Renaming binaries with tag name: $TAG"
mv flightplot.jar.zip $JAR_NAME
mv flightplot.app.zip $APP_NAME

# update release links
HTML="
<!doctype html>
<html>
    <head>
        <meta charset='utf-8'>
        <title>FlightPlot latest release</title>
    </head>
    <body>
        <ul>
            <li>OS X App: <a href='$APP_NAME'>$APP_NAME</a></li>
            <li>Any other OS: <a href='$JAR_NAME'>$JAR_NAME</a></li>
    	</ul>
    </body>
</html>
"
echo $HTML > latest.html
```

dagar, 2015-01-16:
Thanks @AndreasAntener. I considered generating the index like this, but also need to expand the list of all historical builds each time. That seemed like a pain in html.

dagar, 2015-01-16:
@LorenzMeier Yes, I can update flow the same way. This s3 stuff could be shared between Firmware and Flow and probably doesn't belong in the Firmware repo anyway (list.html is gpl). Do you want to create a new repo for it under the PX4 organization?

LorenzMeier, 2015-01-18:
@dagar I missed out on the new repo proposal. What would be stored in it? Is the idea to have it deploy the index file to the AWS bucket whenever its updated? I'd like to avoid super complicated logic in the Firmware and Flow Travis files though - if its just a little code duplication I wouldn't mind if the AWS upload is copy & paste and the CI repo just holds the index file and scripts to view what the other repos upload.

dagar, 2015-01-18:
@LorenzMeier Two minor issues.
-The little script that generates a directory listing for s3 is GPL3. Obviously it's not actually used in the firmware, but is it a problem to include it in this repo?
https://github.com/PX4/Firmware/blob/s3deploy/Tools/s3/list.html

-The Tools/s3 directory would be duplicated in Firmware and Flow. https://github.com/PX4/Firmware/tree/s3deploy/Tools/s3

LorenzMeier, 2015-01-18:
I'm sold. I've created this repo:
https://github.com/px4/ci-tools

And given you admin access. I think its best if you structure it as you prefer and then send PRs to add to PX4/Firmware and PX4/Flow whatever is missing. Many thanks!

commit 275a9df9 (2015-01-21): fixes #1621 - travis after_script upload all Images to s3 with build status and index
closed 2015-01-21 by dagar