PX4/PX4-Autopilot #1624: Commander state machine should allow switch to auto only if navigator is running

thomasgubler, 2015-01-09:
The implementation should make sure that the check not only works with the default navigator but also allow downstream users to use their own navigator app. One idea is to define a topic which any navigator app needs to publish (mission_result?)

We could introduce such checks also for other state changes to protect against startup-script errors.

thomasgubler, 2015-01-11:
The mission_result topic is probably not suited for this. On publication it triggers mavlink data transmission.

bansiesta, 2015-01-11:
You could change it so that mavlink transmission depends on the values of the fields.

Zefz, 2015-01-28:
What about waypoint_setpoint_triplet? If the commander has received at least one waypoint triplet, then it knows that there is an navigator app running somewhere publishing setpoints. If this is true, then vehicle_status could have a field called ```navigator_available = true``` which determines if we are allowed to switch to auto modes.

thomasgubler, 2015-01-28:
The mavlink app also publishes to waypoint_setpoint_triplet in offboard control mode (circumventing the navigator). So this is not really a good check for the navigator availability.

LorenzMeier, 2015-06-14:
I'm just merging a PR which resolves this. Its ok to have mission_result ending up in a MAVLink transmission, after all the commander and the GCS need to know the mission state.

closed 2015-06-14 by LorenzMeier