PX4/PX4-Autopilot #1644: Reduce boiler-plate code for shell handling

LorenzMeier, 2015-01-12:
We should reduce the boiler-plate code for shell handling (start, stop, status calls) by introducing a small helper class which takes an array of callbacks or similar.

@dagar @DrTon @bansiesta Anyone interested?

DrTon, 2015-01-12:
Will think about this, I had this idea for a long time :)

dagar, 2017-04-11:
Do you think we still need this? It could be PX4 base classes for a typical module or driver including variations with tasks or running in the work queue. It could also include param and mavlink subscribe and update boilerplate.

bkueng, 2017-04-12:
Yes something simple and thread-safe would be helpful. I've fixed and made quite a few bugs in different modules related to that already. It's currently not easy to get it right. Also with respect to calling `orb_subscribe` and others from the correct task.

LorenzMeier, 2017-08-20:
@bkueng Did enough on this to be worth closing it.

closed 2017-08-20 by LorenzMeier