PX4/PX4-Autopilot #1648: Increase coverage of hardware tests

LorenzMeier, 2015-01-12:
@dagar The board now executes the px4fmu-v2_test binary on each PR / commit. It would be great to extend the coverage by adding more tests to ROMFS/px4fmu-v2_test/../rcS. A string of "TESTS FAILED" will make the check logic fail the CI test.

[image]

LorenzMeier, 2015-02-15:
@dagar I did a bunch in the recent tests. Interested in DroidPlanner / Android GCS support?

closed 2015-02-15 by LorenzMeier

dagar, 2015-02-17:
@LorenzMeier I'm very interested in Android GCS, but I'm not sure where PX4 fits in with Tower (aka droidplanner 3) and 3dr services. Is QGC on Android the long term plan?

I'm going to try and tackle PR #1492 this weekend (and verify on my new hex).