PX4/PX4-Autopilot #1649: GCC 4.8.4 Issues - Startup failure and Script Variable Resets

sjwilks, 2015-01-14:
Since switching from GCC 4.7 to 4.8.4 there is an issue with variables in the nsh startup scripts being reset causing script and startup failure. Additionally, at least on PX4 FMU/IO setups the system will not boot correctly if the RC is switched on before booting.

- Script Variables
At some point during script execution a limit of some kind seems to be reached (at least that is how it behaves) and when a variable is defined a previously defined one will be reset. A workaround for the moment seems to be resetting unused variables.

- Startup with RC On
Currently only tested with a PX4 but the above issue with script variables is triggered when powering on when the RC is already on. Required variables get reset and the script fails to complete. If the RC is left off during startup everything starts and the RC can then be turned on without any issues.

- Workarounds
As mentioned for script variables, unsetting variables no longer needed can help but is only a short term solution. What prevents the RC issues, which in turn prevents variable resetting, is placing a 2 second sleep in the rcS script right after the "commander start" line. It appears that something in the commander is clobbering something while it completes in parallel (thread?). This needs closer investigation. 

Of course the other "workaround" is to stay with 4.7 for now ;)

sjwilks, 2015-01-14:
Testing with the latest NuttX might be worthwhile.

Zefz, 2015-01-14:
Ah yes. This has been messing with my head a lot. Good to have someone else confirm this problem :+1: . I guess I will just have to revert back to 4.7 for now!

LorenzMeier, 2015-01-21:
Might want to re-confirm with master. I think we got a few resource constraints there now under control, and are continuing to fix access and string stuff.

LorenzMeier, 2015-06-12:
I presume this is resolved?

closed 2015-06-14 by LorenzMeier