PX4/PX4-Autopilot #1661: Add more robust sensor calibration

hauptmech, 2015-01-18:
I do not have deep experience with the pixhawk system. Please correct me where it's clear my assumption or knowledge is wrong.

I was just looking at saving sensor device id info with accel and gyro calibrations to validate the calibrations on startup. I am concerned that this is a dead-end approach.

Storing the mag devid captures only a subset of (user induced) failure modes. Examples of failures that can still happen include loading calibration parameters from one pixhawk to another with the same hardware layout, and manually changing any of the calibration parameters other than _ID by hand. I'm just showing some failure modes, not suggesting these specific ones should be solved.

Ultimately we want to make sure that a calibration is valid. Any given calibration is specific to a unique sensor on a unique px4. Thus each calibration needs this info to accompany it.

px4 uniqueness can be verified via the STM32 device ID http://www.st.com/web/en/resource/technical/document/reference_manual/CD00171190.pdf (30.2
Unique device ID register)

I have not checked the details of all the sensors. I know that at least one of the sensors does not have a unique id (hmc5883). For board mounted sensors, the devid (bus type, bus number, device type) is not likely to change and can probably be used in lieu. For offboard sensors with no unique identifier, it might be possible to use a measurement signature for identity.

I think that calibration results should probably not be stored in /eeprom/parameters...

I think that all sensors should have their own calibration stored, rather than the current system of having the dominant sensor of a given type having it's calibration stored.

With all sensors having a calibration, Automatic failover, or redundant sensor approaches become available. 

Any thought or comments on this?

LorenzMeier, 2015-01-18:
The ```ver``` shell command provides access to the chip ID, so you can look up the code to access it in systemcmds/ver.

It makes sense to map the calibration to an individual sensor. However, the reason to not instantiate the parameters (and no matter how you conceptionally "store the calibration", it will end up using the FRAM storage requiring the use of the param interface) in the sensor driver was to not link the param storage with the sensor driver, as the two are independent blocks.

Wether this separation is important or not is certainly something worth discussing.

It would not require an architectural change to generalize the calibration to calibrate all sensors at once, and to store N calibration sets (identified by devid + device hash).

I guess the important factor is to wrap it all nicely in a high-level API and make sure it scales. Note that we're porting the Firmware currently to other high-end systems where sensor drivers and parameter storage are completely different, and the calibration routine might be running on Linux while the sensor driver runs on an RTOS. The current architecture had this separation between higher level and lower level in mind and ensured a sensor driver is completely stateless.

hauptmech, 2015-01-18:
I agree that the driver code and calibration data should not be collocated (I did not mean to imply that above). 

I'm not sure an all sensors at once calibration is necessary, maybe not even desirable. Each sensor having it's own calibration data would be sufficient I think.

Are you suggesting a common api for calibration storage? Or something else?

Do you guys have a clear idea about where you want the transition between raw data and conditioned/fused data in the upcoming systems? It looks like right now conditioning is tightly coupled with the drivers and I don't remember seeing a raw value passthrough.

LorenzMeier, 2015-01-18:
You want to store the calibration data at runtime in the driver because the driver runs in interrupt context or on a separate RTOS. You want / need to do filtering or integration there, as you induce otherwise insane amounts of CPU load for context switches at 1KHz *per driver*. A scalable design pattern is:

- Calibrate the sensor in the application level code by resetting its offsets and scaling (like we do it now)
- Store the data per device in the application level
- Load the calibration data on sensor startup
- Let the driver apply the calibration and pre-filter / integrate at high rate, offloading context switch load from the application level processor

The *only* change required vs. the current state would be:

- Introduce MAG0_SCALE_X, MAG1_SCALE_X, MAG0_ID, MAG1_ID, etc. params
- Calibrate all sensors in the current routine (else we can't fall back to a different one!)
- Apply the calibration to all sensors, by matching MAG0_ID, MAG1_ID and the /dev/magX get_devid() IOCTLs

If you have further cleanup suggestions they ware warmly welcome. But just to arrive at the functionality you wanted, only minor improvements are necessary.

hauptmech, 2015-01-28:
I'm about to make this happen. Please confirm that MAG0_SCALE_X, MAG1_SCALE_X, MAG0_ID, MAG1_ID, etc. are the parameter names you want (versus the current SENS_MAG_XSCALE, etc)

LorenzMeier, 2015-02-15:
I'm closing this since I believe we implemented all the items discussed here.

closed 2015-02-15 by LorenzMeier