PX4/PX4-Autopilot #1668: AUTO / RTL modes not working with the ekf_att_pos_estimator

sjwilks, 2015-01-19:
Attempting to switch to AUTO results in a state change rejection. This means that currently all sub-modes such as RTL will also not currently work (so watch out when testing). It looks like there is at least an issue with setting home correctly.

thomasgubler, 2015-01-19:
We need to check if FW still works on master

sjwilks, 2015-01-19:
Auto for FW is broken in master. I narrowed it down to #1532. 

@kd0aij would you mind taking a look and seeing if you can verify this and then see if you can spot why it prevents switching to auto (FW and MC). Thanks.

thomasgubler, 2015-01-19:
Since the home position seems to be the problem I suspect that one of these checks here fails:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L734 https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L739. Because #1532 changes the altitude estimate I suspect it is the later one.

kd0aij, 2015-01-19:
This would be because the diagnostic code (labeled FIXME:) at lines 1539-1544 didn't get removed. That code alters eph and epv and would definitely cause the checks to fail. Sorry for not catching that before the merge.

LorenzMeier, 2015-01-19:
Thanks Mark. Fixed via fa4d09018527676d3803582a432df8e25c8e5010 and pushed as hot fix to master.

@sjwilks Let us know if that fixes things. A handheld test outdoors is probably already enough to confirm the fix.

sjwilks, 2015-01-19:
Great. I'll test it first thing in the morning.

sjwilks, 2015-01-20:
Retested on the bench using GPS_FAKE and switching to AUTO now works.

LorenzMeier, 2015-01-20:
Cool. Closing for now, let us know if any issue persists.

closed 2015-01-20 by LorenzMeier