PX4/PX4-Autopilot #1682: UAVCAN clean target still not operational

LorenzMeier, 2015-01-21:
Just got this, and for me its a bit head vs. wall - I've tried a bunch of things on this already. I presume its a wrong path in the dependency files, and probably stems from how uavcan hooks into the build system. Neither clean / distclean are currently operational.

@davids5 any insights?

```
make[2]: *** No rule to make target `/Users/lorenzmeier/src/Firmware/Build/px4fmu-v2_default.build/nuttx-export/include/nuttx/mm/shm.h', needed by `/Users/lorenzmeier/src/Firmware/src/lib//uavcan/libuavcan//src/driver/uc_can.cpp.o'.  Stop.
make[2]: *** Waiting for unfinished jobs....
CXX:     /Users/lorenzmeier/src/Firmware/src/modules/uavcan/sensors/baro.cpp
PRELINK: /Users/lorenzmeier/src/Firmware/Build/px4fmu-v2_default.build//Users/lorenzmeier/src/Firmware/src/modules/systemlib/module.pre.o
PRELINK: /Users/lorenzmeier/src/Firmware/Build/px4fmu-v2_default.build//Users/lorenzmeier/src/Firmware/src/modules/mavlink/module.pre.o
make[1]: *** [/Users/lorenzmeier/src/Firmware/Build/px4fmu-v2_default.build//Users/lorenzmeier/src/Firmware/src/modules/uavcan/module.pre.o] Error 2
make[1]: *** Waiting for unfinished jobs....
PRELINK: /Users/lorenzmeier/src/Firmware/Build/px4fmu-v2_default.build//Users/lorenzmeier/src/Firmware/src/modules/ekf_att_pos_estimator/module.pre.o
make: *** [/Users/lorenzmeier/src/Firmware/Build/px4fmu-v2_default.build/firmware.px4] Error 2
```

pavel-kirienko, 2015-01-21:
For reference: https://github.com/PX4/Firmware/issues/1404 (see comments)

LorenzMeier, 2015-01-22:
Ok, I think I've put this finally to rest:
https://github.com/UAVCAN/uavcan/commit/7719f7692aba67f01b6321773bb7be13f23d2f68

https://github.com/PX4/Firmware/commit/7faef870c8201696372661e063336069d91123b9

The paths were containing double-slashes, which broke dependency generation. @davids5 This is already applied to nuttx_bringup, so you should not see this again when switching between nuttx_bringup and master.

pavel-kirienko, 2015-01-24:
I guess this is fixed now.

pavel-kirienko, 2015-01-28:
Nope, still there.

LorenzMeier, 2015-06-12:
Fixed by @davids5 .

closed 2015-06-12 by LorenzMeier