PX4/PX4-Autopilot #1683: Gazebo simulator: investigate why the outdoor world does not launch on ubuntu

thomasgubler, 2015-01-22:
The outdoor_world currently seems to work on arch linux only. empty_world works fine on all systems.

Symptoms: gazebo starts but can't find some model files, shows a black screen

This issue is in this repo because my guess is that something in the setup script is wrong.

thomasgubler, 2015-01-22:
For the house_world some models are not found as well. Probably a PATH_XYZ variable is not set

thomasgubler, 2015-01-22:
both the house_world and the outdoor_world set the `GAZEBO_MODEL_PATH` to use the *speical* models. It seems that this is not working on ubuntu or the path is wrong.

mhkabir, 2015-01-23:
I've been able to reproduce this too on 14.04 Indigo.

thomasgubler, 2015-02-07:
I can reproduce this now as well with archlinux and gazebo 5 (I felt adventurous...)

vageeshb, 2015-03-24:
As Thomas mentioned, these worlds use `GAZEBO_MODEL_PATH` to use the models. The current `outdoor_world.launch` defines the `GAZEBO_MODEL_PATH` with the line
```
<env name="GAZEBO_MODEL_PATH" value="$(find drcsim_model_resources)/gazebo_models"/>
```
There are 2 fixes/workaround that has worked for me:
- Removing the above line from the launch file and sourcing the variable value into bash from `drcim` package itself
```bash
$ source /usr/share/drcsim/setup.sh
```

- The models are under the 'environments' folder, therefore replacing the above line with
```
<env name="GAZEBO_MODEL_PATH" value="$(find drcsim_model_resources)/gazebo_models/environments"/>
```

vikrantvm, 2015-04-14:
A workaround I found was calling the launch file in the rotors_gazebo folder(iris_house_world.launch). I moved the other launch includes in this launch file. I am able to fly the copter with a joystick. I hope this is useful. The launch file looks like this: 

            <launch>
        <arg name="enable_logging" default="false"/>
        <arg name="enable_ground_truth" default="false"/>
        <arg name="log_file" default="iris"/>
        <arg name="headless" default="false"/>
        <arg name="gui" default="true"/>
        <arg name="ns" default="iris"/>

        <env name="GAZEBO_MODEL_PATH" value="$(find rotors_gazebo)/models"/>
        <include file="$(find gazebo_ros)/launch/empty_world.launch">
          <arg name="world_name" value="$(find rotors_gazebo)/worlds/house.world"/>
          
          <arg name="headless" value="$(arg headless)"/>
          <arg name="gui" value="$(arg gui)"/>
        </include>
        <include file="$(find rotors_gazebo)/launch/spawn_iris.launch">
          <arg name="model" value="$(find rotors_description)/urdf/iris_base.xacro" />
          <arg name="enable_logging" value="$(arg enable_logging)" />
          <arg name="enable_ground_truth" value="$(arg enable_ground_truth)" />
          <arg name="log_file" value="$(arg log_file)"/>
        </include>
        <include file="$(find px4)/launch/iris.launch">
      	<arg name="ns" value="$(arg ns)"/>
          </include>
      </launch>

LorenzMeier, 2016-03-13:
Not relevant.

closed 2016-03-13 by LorenzMeier