PX4/PX4-Autopilot #1700: Inspect side-effects of disabling stack checking

LorenzMeier, 2015-01-26:
This includes #1691. @davids5 @sjwilks These are the combined symptoms of disabling stack checking:

  * V1: microSD mount fails
  * V3: USB serial link (NuttX) is started, but never registers with the host (device doesn't list under /dev/tty.usb*)

I'm not 100% sure yet if this is timing or if we're missing something, but we clearly don't have everything under control yet.

LorenzMeier, 2015-01-26:
@davids5 Ok, I've put the USB one to rest. Fixed via 18d756d. So only the microSD mount fail remains, and I wouldn't be terribly surprised if its timing again there.

davids5, 2015-01-26:
@LorenzMeier Craig has confirmed that there will be HW heading my way at some point. I will dig into this bug then,

LorenzMeier, 2015-01-31:
@sjwilks Could you try disabling the ms5611 driver startup? I'm wondering if the new SPI variant messes with the SD bus and the timing change made it visible.

sjwilks, 2015-01-31:
While trying to workout why my F330 (PX4 V1) was going nuts since yesterday
I noticed that the SD card seems to have died completely. This probably
means my tests in the last two days weren't reliable. I have just replaced
the card, re-flashed master (via QGC) and I can read from the SD card. So
it looks like #1691 did indeed help. Prior to that I could constantly
read/not read based on the stack check setting so I think the SD failure
only just began.

LorenzMeier, 2015-03-03:
Fixed.

closed 2015-03-03 by LorenzMeier