PX4/PX4-Autopilot #1706: nuttx-bringup: multicopter: motors won't spin

thomasgubler, 2015-01-26:
@davids5 This needs more investigation. I flashed the nuttx-bringup branch onto a pixhawk on a multicopter. I can arm but the motors won't spin if I throttle up. It seems that the controller is running correctly and publishing meaningful data to the actuator controls uorb topic.

Is this a known issue or do you wan't me to investigate further? (E.g. look at pwm output, mixer, ...)

Zefz, 2015-01-26:
Tried having your RC *off* when powering your multicopter? Only turn on the RC *after* you hear the confirmation sound of the ESC power-up.

davids5, 2015-01-26:
@thomasgubler This is the first report of it. Anything you can do to isolate it would be very much appreciated.

sjwilks, 2015-01-26:
@Zefz that's the same symptom I had in #1649. Does a sleep after starting commander help?

davids5, 2015-01-26:
@thomasgubler I just pushed a fix that deals with the communications with the pio. I can imagine that this issue you're seeing might have been related. Would you please retest with the latest nuttn_bringup  9dcd7ef35e0beb8f30f3901759210f14a64287ff

thomasgubler, 2015-01-27:
@davids5 unfortunately that didn't help. I will investigate this a bit more...

thomasgubler, 2015-01-27:
The issue is that the mixer is not loaded correctly because some variables ate not expanded in the rc.interface script. Fixed in #1708

PR #1708 (2015-01-27): rc.interface: expand mixer variables
@davids5 @LorenzMeier I am not sure if the behavior of nsh/NuttX changed. If I remember correctly we had our own string concatenation implementation and I guess this is now the standard NuttX version.

Fixes #1706
closed 2015-01-27 by thomasgubler