PX4/PX4-Autopilot #1720: nuttx-bringup: make archives modifies NuttX submodule

thomasgubler, 2015-01-28:
Before `make archives`:
```
/NuttX git:remotes/origin/upstream git status                    
HEAD detached at 69283d3
nothing to commit, working directory clean
```

After `make archives`:
```
/NuttX git:remotes/origin/upstream git status                                                                                                                                                      ✱ ◼
HEAD detached at 69283d3
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   nuttx/include/signal.h
	modified:   nuttx/include/stdio.h
	modified:   nuttx/include/stdlib.h
	modified:   nuttx/include/sys/stat.h

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	nuttx/.patchedpx4common
	nuttx/arch/arm/include/math.h

no changes added to commit (use "git add" and/or "git commit -a")
```

[Output of git diff](http://pastebin.com/cbEYvYFF)

davids5, 2015-01-28:
@thomasgubler The nuttx_bringup build supports build time patching of the things we can not get into nuttx upstream (PX4 specific requirements that may be as simple as gcc pragmas)  

The make targets nuttxpatches and cleannuttxpatches apply and remove the patches.

Currently 
 archives depends on nuttxpatches 
 distclean depends on cleannuttxpatches

So a make distclean (or cleannuttxpatches) will bring NuttX back to unmodified state.

Would you like this to work differently?

thomasgubler, 2015-01-28:
The problem is that the NuttX submodule shows up as modified when running git status in the Firmware repo. So you can never be sure if you have a clean version of NuttX. Are the patches required for the Firmware build? Would it maybe make sense to run cleannuttxpatches automatically after a archives build?

davids5, 2015-01-28:
For the developer whose primary focus is on PX4 Firmware Yes, it would make sense to run cleannuttxpatches automatically after a archives build. So I totally agree I think that should be the default.

For developer whose primary focus is on Nuttx development for PX4. I would like to see changes persist.

When I added the patching I was thinking there were common (one time patches) and possible per hw targets patches. (I still have not seen a case for the per hw patching yet.)

I would like to to see some sort if a "mode" setting that would allow one to focus on a HW platform and be able to build the PX4 Firmware on top of Nuttx without exporting the archive. This would allow the developer to incrementally build Nuttx and save a lot of time. 

We need some makefile magic - I would appreciate any suggestions and help implementing same.

thomasgubler, 2015-01-28:
> For developer whose primary focus is on Nuttx development for PX4. I would like to see changes persist.

You want to see the changes persist but do you really want them to show up as a change in git? Ideally git would only show changes if you did a change to the patch? (which would again be ok for the developer with px4 firmware focus)

davids5, 2015-01-28:
We should probably have a Skype discussion on this. I am david_s5

davids5, 2016-05-17:
Cmake made this a non-issue

closed 2016-05-17 by davids5