PX4/PX4-Autopilot #1730: Config parsing / checking tool: Write unit test

LorenzMeier, 2015-01-30:
We found that different shells can lead to different testing results. As this influences the build config, we need to write an unit test which fails the build if it doesn't find the expected values in a test file.

Test cases should include:

  * Whitespace differences
  * Disabled values as commented (# XYZ is not set) and =n
  * Different line endings (Unix, Windows)

A failing unit test should fail the complete build immediately and print an error message providing clear instructions about the issue and contact points.

nuttx, 2015-01-31:
There is a tool at nuttx/cmpconfig.c that I used to see if there are meaningful differences between configurations.  It would handle your whitespace or line ending differences, but will certainly tell you if there are logic differences between any two configurations.

nuttx, 2015-01-31:
Ooop.. nuttx/tools/cmpconfig.c

closed 2016-05-22 by LorenzMeier