PX4/PX4-Autopilot #1732: PWM min limit 1300

AndreasAntener, 2015-01-31:
I recently tried to set the minimum PWM value to 1500 but it always got "stuck" at 1300. There's a max min limit defined here: https://github.com/PX4/Firmware/blob/master/src/drivers/drv_pwm_output.h#L80

Why is this limit necessary?

The reason I tried to set min to 1500 is the servo for a parachute release. It needs 2 states: "normal" and failsafe. For the normal position I set "disarmed" and "
min" PWM values. I know I could also adjust the servo in this case, but I'm wondering about the max min limit anyway..

bansiesta, 2015-02-04:
This max min limit was supposed to prevent users from entering too high minimum values, so to ensure "safe" values. I see that for your use case this doesn't really make sense. I guess you could change it to 1500.

AndreasAntener, 2015-02-04:
Thanks, I feel it's not necessary at all, same for the "min. max. limit". What's "safe"? ;)
It should only check if the values make sense regarding the other values (e.g. min limit lower than max limit). Maybe it also does that already, didn't check.

bansiesta, 2015-02-07:
Yes, it's not "safe" anyway :)

LorenzMeier, 2015-02-15:
@AndreasAntener PR welcome.

LorenzMeier, 2015-03-27:
I changed this to 1600 for max low and 1400 for min high so this should be resolved now.

closed 2015-03-27 by LorenzMeier