PX4/PX4-Autopilot #1745: multiplatform/ros sitl: add mavros interface

thomasgubler, 2015-02-05:


mhkabir, 2015-02-10:
@thomasgubler  Would adding mavros as a whole actually benefit us? Perhaps it would be better if we  adapted a inteface from the mavros communications core ("libMavConn")
https://github.com/mavlink/mavros/tree/master/libmavconn

thomasgubler, 2015-02-10:
The idea is to have a drop in replacement so that the sitl really acts like
the real system. I have to look at libmavconn though.

thomasgubler, 2015-02-10:
ok it probably makes sense to use libmavconn directly on the sitl side. The plan is something like this:
```
any custom ros app <--> mavros <--> |UDP| <--> [new node with libmavconn] <--> controller
```

The lefthand side is the part which goes on the companion computer on the real system

thomasgubler, 2015-02-10:
I added a (skeleton) mavlink node here: https://github.com/PX4/Firmware/tree/ros_mavlink

PR #1793 (2015-02-28): ROS SITL: offboard setpoints
# What's new
  * mavlink dummy node which forwards a small subset of messages via udp to a mavros node
  * Support for local position and attittude offboard setpoints (XXX_target mavlink messages) in ROS SITL
  * Small demo node for offboard position control
  * launch files which launch the demo node and setup mavros: `roslaunch px4 gazebo_ardrone_empty_world_offboard_demo.launch`

Here is a small overview:
```
(Gazebo <--> ) [demo_positon_setpoint_generator node] <--> [mavros] <--> UDP <--> [mavlink node] <--> [controller nodes] <--> Gazebo
```

If you use your o [... cut]
closed 2015-02-28 by thomasgubler