PX4/PX4-Autopilot #1750: SITL installation script: drcsim installation fails on ubuntu

thomasgubler, 2015-02-05:
@AndreasAntener  Can you maybe give this a test when you run the installation anyway for some reason? It's not high priority though.

thomasgubler, 2015-02-05:
From @mverzilli:

>1. The px4_ros_installation_ubuntu.sh works great except for the last line: sudo apt-get -y install drcsim. It gives “Unable to locate package drcsim”. 

>By inspecting the corresponding Packages list, I can see that indeed drcsim isn’t there (http://packages.osrfoundation.org/drc/ubuntu/dists/trusty/main/binary-i386/Packages). 

>But Gazebo’s own installation instructions do suggest installing drcsim just like your installation script does, so I’m a bit clueless as to what happened to that package. My guess is it either got subsumed but something bigger (maybe gazebo itself?) or it got renamed.

mverzilli, 2015-02-05:
Thanks for taking care of this! I've got some more info now: 

In https://bitbucket.org/osrf/drcsim/wiki/DRC/ChangeLog, for the last version they mention: 

````
Debian distributions moved to new server (http://old.gazebosim.org/assets/distributions/).
````

There's a bunch of .deb packages over there for different versions of drcsim, so I could try dpkg installing the appropriate one.

AndreasAntener, 2015-02-05:
I just noticed that the repository for i386 doesn't have drcsim but the one for amd64 does. @mverzilli are you installing on a 32bit system?

Just found this:
"Note: as of drcsim 2.2.0, binary packages are available for 64-bit systems only. A 32-bit version is still available as source, but 64-bit is required to run the BDI walking controller."
from [here](http://gazebosim.org/tutorials?tut=drcsim_install)

I'm not sure what this means for us, but ATM we actually don't need drcsim right? @thomasgubler ;)

AndreasAntener, 2015-02-05:
Otherwise, the setup still works for me (docker container built successfully)

mverzilli, 2015-02-05:
Indeed. I was attempting all of this on a 32-bit Ubuntu 14.04 VirtualBox image. It actually was an accident, I intended to download a 64-bit image :/. I'll give it a try on a 64-bit VM and see if it works (it looks like it will :)). Thanks for the quick response, pretty impressive!

AndreasAntener, 2015-02-05:
Great :)
But just so you know: I'm having trouble compiling the whole thing at the moment. There might be really a problem there somewhere.

AndreasAntener, 2015-02-05:
Ok, it seems that compilation for me failed because my VM didn't have enough memory (only 1GB). Got it working with 4GB.
Looking forward to your results ;)

mverzilli, 2015-02-09:
Hey! Sorry for the delay, it took me a while to download a 64-bit image and I switched to work on other things. So, I followed the tutorial again and this time everything went smoothly :). Now I'm blocked because it seems I don't have access to a couple of repos:

https://github.com/PX4/euroc_simulator.git
https://github.com/PX4/mav_comm.git

Could you grant me access? 

By the way, there's a broken reference to the script px4_workspace_setup.sh at https://pixhawk.org/dev/ros/sitl (I still grabbed it from https://github.com/PX4/Firmware/blob/master/Tools/ros/px4_workspace_setup.sh).

thomasgubler, 2015-02-09:
You already have access. See my email from Feb 5. You need to accept the
invitation.

-Thomas

thomasgubler, 2015-02-09:
closing since the original issue was resolved/present only on i368

closed 2015-02-09 by thomasgubler

mverzilli, 2015-02-09:
@thomasgubler @AndreasAntener thanks for your help! I was able to build and run it.