PX4/PX4-Autopilot #1759: MC attitude control: gains falsly initialized?

tumbili, 2015-02-08:
Connected battery and noticed that elvons did only respond slightly to my TX pitch commands (used the Caipirinha VTOL). Logs confirmed that ATTC.Pitch was at most 8% of the max value although pitch rate error was huge. Controller gains were untouched.
Unplugged and replugged the battery and all of a sudden everything was ok again.
Log is here: http://dash.oznet.ch/view/ktMUxGX8espdDHWvma49hD
Please download and analyse with FlightPlot, LOG Muncher won't show!

LorenzMeier, 2015-02-08:
@tumbili It might very well be fixed in 0fd11b7. Let me know if rebasing on master fixes it for you.

tumbili, 2015-02-08:
I will have to test it over the next days, the problem only occured rarely.

closed 2015-04-29 by tumbili

kd0aij, 2015-04-29:
What was the reason for closing?

tumbili, 2015-04-29:
@kd0aij After all this time I have never witnessed this problem again.