PX4/PX4-Autopilot #1767: Parameter set needs to be version stamped

DonLakeFlyer, 2015-02-10:
There needs to be a parameter which is the version for the parameter set. Whenever the parameter version gets bumped in the source it indicates a breaking change to the parameter set.

Why: Recently the parameters associated with sensor calibration were all changed. This broke a pile of QGroundControl functionality. QGC needs to know when it is attempting to work against a parameter set that it does not yet understand. Also if you reload a saved parameter file QGC needs to be able to verify that the set of parameters in that file are valid for the firmware on the board.

DonLakeFlyer, 2015-02-10:
This is fairly high priority since as it stands now I think the parameter set between the development and stable firmware version may actually be different versions. If so, that leaves QGC in a bad state as to what to do and how to run against each of these. Without the version stamp it doesn't have much chance of working correctly. It will just confusingly fail for users depending on firmware version.

LorenzMeier, 2015-02-13:
Do you have a preferred implementation method? E.g. a version param? Is there a good pattern from APM we could adopt?

DonLakeFlyer, 2015-02-13:
APM just has a simple integer version parameter (read-only). They bump it up whenever a breaking change is made:
- Parameter delete/rename
- Semantic change to parameter

LorenzMeier, 2015-06-14:
We have this now I believe?

closed 2015-06-14 by LorenzMeier