PX4/PX4-Autopilot #1787: UAVCAN: non-zero motor output while disarmed

hsteinhaus, 2015-02-14:
The native Pixhawk stack seems to send non-zero throttle values to UAVCAN ESCs while the vehicle is disarmed (no difference if safety is on or off). 

I think the minimum value from mixer definition is used to inherit the actual disarmed throttle value - I have set 1000=10% (4th value of the MC quad mixer) for the "uavcan status" dump  below", as the printing code is displaying a range from -1000...1000.

uavcan status
ESC actuators control groups: sub: 1 / req: 1 / fds: 3
ESC mixer: OK
ESC output: -800 -800 -800 -800 0 0 0 0 
ESC Status:
Addr    V       A       Temp    Setpt   RPM     Err
10      11.36   -0.01   NaN     0.00    0       0
11      11.38   0.00    NaN     0.00    0       0
12      11.38   -0.01   NaN     0.00    0       0
13      11.38   -0.01   NaN     0.00    0       0
0       0.00    0.00    0.00    0.00    0       0
0       0.00    0.00    0.00    0.00    0       0
0       0.00    0.00    0.00    0.00    0       0
0       0.00    0.00    0.00    0.00    0       0

LorenzMeier, 2015-02-15:
Yes, that is according to specs. Subsystems should send their outputs regardless of the armed state. Makes system checks much easier. So we're missing the logic in the UAVCAN bridge to nail those to zero?

hsteinhaus, 2015-02-16:
After some code reading I think I understand the problem better now: The armed topic is employed to suppress the throttle output in the ESC driver, not the uavcan bridge itself. That is why the "uavcan status" output (which is handled by the bridge) does not print the true values sent to the ESC. 

My initial problem (no motor startup possible with minimum throttle > 10%) is obviously caused by the startup logic of the ESCs, that prevents a startup with a higher throttle level than 10%. If a higher idle power than 10% is necessary for aerodynamic reasons, the ESCs refuse to arm. May be some kind of throttle ramping is needed here?

LorenzMeier, 2015-02-16:
Throttle ramp sounds like it makes a lot of sense. The PX4IO driver employs a 3-5 seconds ramp from disarmed to current throttle / PWM, and we could do the same there. It has the nice side-effect that an accidental arming results in minimal throttle first and usually people are fast enough to get their fingers out of the way within the first 1-2 seconds.

closed 2015-10-14 by LorenzMeier