PX4/PX4-Autopilot #1789: Multirotor auto takeoff without takeoff waypoint can fail

sjwilks, 2015-02-14:
If an auto takeoff is performed and the next waypoint in the mission is not a takeoff waypoint the multirotor tries to fly directly to the waypoint on takeoff rather than first climbing vertically to a safe height. This usually results in a failed launch as the multirotor tilts towards the waypoint.

thomasgubler, 2015-02-15:
Makes me wonder if we should get rid of the takeoff wp altogether? (At least for multicopters).

sjwilks, 2015-02-15:
True. I can't think of anything at the moment that a takeoff waypoint would
do differently now or in the future.

tumbili, 2015-04-14:
@sjwilks @thomasgubler @LorenzMeier Is this an error?
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L452
I assume that altitudes are expressed in the NED frame? This would explain why the take-off to a save altitude is skipped.

LorenzMeier, 2015-04-14:
See here:
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L438-L441

What is intriguing though is that in that line we're using the acceptance radius, which is param NAV_ACC_RAD, which defaults to 2 meters if the vehicle was set up using auto-config steps.

The radius is the error allowed in 3D, so at the very least we need to take e.g. 1/3 to represent that this is now just in Z. Since this is an initial check I would even go as far to say that the radius should be deleted at this particular position.

LorenzMeier, 2015-04-14:
@sjwilks You will want to check our NAV_ACC_RAD value.

thomasgubler, 2015-04-15:
I would also vote for removing the radius in this case.

LorenzMeier, 2015-04-15:
@sjwilks Let us know if it persists, assuming this to be closed for now.

commit c9c1c9d5 (2015-04-15): Fix #1789 - takeoff sideways on beginning of missions
closed 2015-04-15 by LorenzMeier