PX4/PX4-Autopilot #1806: Architechtural changes for Linux port (multiplatform)

mhkabir, 2015-02-17:
@thomasgubler @LorenzMeier 
The normal apps which run on NuttX are excellent for porting to Linux since they already use scheduling, process priorities, and other POSIX style code.

The multiplatform API apps are ported with simulation in mind and Thomas separated the code into start_nuttx for NuttX, and remove the scheduling abilities like task_spawn from the ROS version.
Since we need proper scheduling, accurate timing (from a monotonic clock) in the Linux port, I propose to make a unified start function for Linux and NuttX which will only differentiate in calling ros::init(), pub-sub init etc. depending on the platform. Scheduling will work accurately on relatime flight systems, and for running just simulation, it just won't be hard realtime.

This is essential for the Linux port. Also, architecture wise, I saw that it'd save a lot of time and effort of  rewriting drivers if we remain with the main codebase. The driver re-work to run on multiple platforms as Lorenz was saying would be very useful here.

Currently I have to get the commander, mixer, and some more apps over to multiplatform API. The "dummies" as implemented currently can be removed I think, and made "bridge" nodelets to simply translate between ROS and PX4.

Regarding frame conventions in the port, I'm not sure what to do here. Perhaps run the flightstack internally in NED and add bridge nodes for translating from standard ROS messages like sensor_msgs/IMU and Odometry, etc. allowing us to use external state estimation nodes like ethzasl_msf, etc.

Comments and suggestions are welcome.

mhkabir, 2015-02-17:
@jserv 
I'd also like to introduce Jim who has agreed to help us with hard realtime in Linux. He has experience with Xenomai dual-RT kernel on several platforms. 
Jim, please raise your concerns and whatever you'd like to say ;)

thomasgubler, 2015-02-17:
> I propose to make a unified start function for Linux and NuttX which will only differentiate in calling ros::init(), pub-sub init etc. depending on the platform. 

The current ros implementation was just the most simple way to get it working without doing over-engineering for the simulation case. Also was mostly following the ROS examples to make it as 'easy' as possible. If we now need something more sophisticated I see no reason to stick to the old pattern.

> Currently I have to get the commander, mixer, and some more apps over to multiplatform API. The "dummies" as implemented currently can be removed I think, and made "bridge" nodelets to simply translate between ROS and PX4.

Are you working on this? Can you please sync with me? :) Yes the plan is to remove the dummy nodes. Those are just minimal implementations to test and showcase the setup.

>Regarding frame conventions in the port, I'm not sure what to do here. Perhaps run the flightstack internally in NED and add bridge nodes for translating from standard ROS messages like sensor_msgs/IMU and Odometry, etc. allowing us to use external state estimation nodes like ethzasl_msf, etc.

I guess because we are writing software for aerial vehicles we should stay with NED standards to make everyones life easier. Translating at the interface to ROS makes sense.

mhkabir, 2015-02-18:
@thomasgubler 
1. As Lorenz would like to collapse the apps back into one, I think this would be an ideal chance to implement the first change, atleast partially. It might need some changes in systemlib, but it should work mostly out of the box. 
2. I am frustratingly busy with exams till the 23rd of March, so am not able to do anything for now. I will be glad to coordinate on this once I'm free.
3. Makes sense.

thomasgubler, 2015-02-18:
> As Lorenz would like to collapse the apps back into one, I think this would be an ideal chance to implement the first change, atleast partially. It might need some changes in systemlib, but it should work mostly out of the box. 

I suggest to first bring the apps back together and then start with the scheduling changes to save us some headaches.

mhkabir, 2015-02-19:
Yep, that's good. 

@LorenzMeier Can you comment on the task vs. pthread question for scheduling ? The flightstack uses (non-POSIX) tasks for everything, while we can only(?) use pthreads in Linux.

LorenzMeier, 2015-02-19:
Pthreads would be a terrible choice. Stick to the ROS standard and use tasks.

mhkabir, 2015-02-19:
@LorenzMeier @thomasgubler 
Regarding timestamps for the whole multiplatform system, what should we use? Ideally I think we should use a monotonic time source within the flightstack and change that to ROS time at the interface. I am sticking to this for the driverlayer atleast.
ROS time is susceptible to change (from NTP sync, etc.) but the boot clock would remain unaffected.

LorenzMeier, 2015-02-19:
@mhkabir In order to have good SITL capabilities, we need to use the ROS time and not the boot time - else all the controller would be unusable in SITL
http://wiki.ros.org/roscpp/Overview/Time

I don't want to bake in ROS parts into the driver layer though, so I agree that the drivers should use the monotonic clock (via normal POSIX calls). You can implement a call to that and wrap it in the px4 time / hrt_absolute_time() calls.

I would also recommend you to have a look at the driver base class, because it is effectively a HAL across systems, and it would make all of the drivers we have portable across operating systems. You just need to re-implement the I2C and SPI classes.
https://github.com/PX4/Firmware/blob/master/src/drivers/device/i2c.cpp#L231-L254

mhkabir, 2015-02-19:
So exactly, that's what I want. We use the monotonic time inside the flight
stack/ middleware.
Where we changeover to ROS standard topics for SITL (or otherwise) we
change it to ROS time and reverse as required . A sort of "bridge" layer I
proposed above, replacing the dummy nodes.

I'll have a look in to the driver base class. I like it. It'd be good to
invest time there NOW and simply write a mpu9250 driver in Firmware. How'd
we differentiate between platforms?

Also in which case, we might totally drop a external driver layer and put
everything inside the main codebase.

vooon, 2015-02-19:
@mhkabir Note: `<chrono>` have monotonic clock abstraction.

LorenzMeier, 2015-07-24:
I'm closing this since this discussion should probably be rebased against the actual implementation in the POSIX port.

closed 2015-07-24 by LorenzMeier