PX4/PX4-Autopilot #1809: Collapse mc_att_control and mc_att_control_multiplatform into one

LorenzMeier, 2015-02-18:
The two apps are drifting apart with the speed of sound (lots of progress, good!), so we need to do this step now before its too late.

mhkabir, 2015-02-18:
@thomasgubler It'd be awesome to start phasing in the changes I was telling you about to maintain compatibility with Linux port.

thomasgubler, 2015-02-18:
I opened https://github.com/PX4/Firmware/issues/1811 to track the merging of the position controller.

mc_att_control_multiplatform and mc_pos_control_multiplatform are both stable at the moment and no changes are planned regarding the API and SITL which would require updating these apps.

Hence, I suggest this approach:
1. Disallow pull requests against mc_att_control and mc_poscontrol
2. Collapse the apps: this basically consists of porting the last few pull requests against mc_att_control/mc_pos_control to the multiplatform apps.
3. rename multiplatform apps to the standard name/remove old apps. --> multiplatform is default on master

If step 2 has to happen fast someone else than me has to do it, sorry.

thomasgubler, 2015-02-18:
If I am not looking at the wrong github history this is still the only thing to be ported:
https://github.com/PX4/Firmware/pull/1741

here is a quick link to the history of mc_att_control:
https://github.com/PX4/Firmware/commits/master/src/modules/mc_att_control

closed 2015-02-24 by thomasgubler