PX4/PX4-Autopilot #1810: Test position / attitude control on master with offboard control logic

LorenzMeier, 2015-02-18:
We need to re-test position / attitude offboard control on master, since we fixed a number of issues after a recent cleanup.

@AndreasAntener @sjwilks Could you contribute there?

LorenzMeier, 2015-02-18:
@thomasgubler Looks like we need to really get rid of the multi platform code, as our testing diverges as well. I didn't expect us to move so fast, but now that we do its critical.

thomasgubler, 2015-02-18:
Anybody who finds time to test: please merge this branch into your test branch because it's on the critical path for sitl offboard https://github.com/PX4/Firmware/pull/1786

aerialhedgehog, 2015-02-18:
I can do a test flight for position control on saturday.  Just so i'm clear on the branches that need to be tested --

- branch off of master
- merge PR #1786 
- test various setpoints from offboard

LorenzMeier, 2015-02-19:
Test both master and the PR you mentioned please.

thomasgubler, 2015-02-19:
Cool! Don't forget to do a bench test first, especially with the PR. So you don't waste time if it doesn't work.

AndreasAntener, 2015-02-20:
We did two offboard test flights today (video in the works :)), one on latest master and one on master merged with #1786 . Both worked fine.

AndreasAntener, 2015-02-20:
Oh, but only position setpoints, not attitude.

Zefz, 2015-02-20:
@AndreasAntener Makes sense. The ```mc_pos_controller``` only enables direct attitude control if ```control_manual_enabled``` is true (which it isn't for offboard control). Does the offboard control directly simulate manual stick control input?

mhkabir, 2015-02-20:
I did some hand-testing and attitude control doesn't work. Thrust + position does. I think @Zefz has the root cause.

AndreasAntener, 2015-02-20:
@Zefz Offboard control directly commands a position setpoint.
Video from our flight: https://www.youtube.com/watch?v=0hdEZQ2MAW8

I can't really comment on the offboard attitude control because I have no clue if it was event there once ;)

aerialhedgehog, 2015-02-22:
Some more flight tests - -
https://www.youtube.com/watch?v=K-pBmei5Uso

Here's a table of results on my end - -
- Firmware/master
  - velocity control - works, moves with set velocity (did not video this)
  - position  control - works, but i can't control how fast it moves to the next set point
- Firmware/offboard (ref PR #1786)
  - velocity control - *does not move*
  - position  control - works, but i can't control how fast it moves to the next set point

It's a bit surprising, if not frightening, to see how fast my hex wants to move to a new position set point.  I was thinking this is governed by the parameter MPC_XY_VEL_MAX, but while walking back from the lake i realized maybe this is governed by MPC_TILTMAX_AIR.  While not relevant to this issue, I think it would be useful to consider introducing offboard control over velocity and acceleration magnitudes.

Additional comment, I didn't have any issues with my motors cutting out, which would close this issue from c_uart_example - -
https://github.com/mavlink/c_uart_interface_example/issues/11

AndreasAntener, 2015-02-22:
@aerialhedgehog We too were "fooled" by the velocity parameters. We tried several flights with different settings but not effect. I just did a test with MPC_TILTMAX_AIR in SITL and it does the trick.

sjwilks, 2015-02-22:
That won't help with vertical velocities though. The only other workaround
is to have intermediate setpoints. So the question is should the MPC
velocity params work or are we supposed to set velocities ourselves? Or
both?

LorenzMeier, 2015-02-22:
The idea behind the original design was that the velocity, if provided and valid, should be the velocity the system is aiming for. I think there is some relaxing logic necessary: The commanded velocity should be min(velocity_command, velocity_pos_controller_output, max_vel_param). If done this way the commanded velocity can be used to control the system response, but if its too large it won't overshoot once it reaches the desired position.

LorenzMeier, 2015-02-22:
This probably needs to be implemented and is currently not yet - anyone? 8).

AndreasAntener, 2015-02-22:
So your saying it would then be possible to set both, velocity and position setpoints at the same time, and the controller would use the velocity setpoint as max velocity?

aerialhedgehog, 2015-02-22:
@AndreasAntener yeah i think this is the idea.  
maybe we need a new issue for this?

closed 2016-03-24 by LorenzMeier