PX4/PX4-Autopilot #1817: Battery internal resistance and no-load voltage estimation

dogmaphobic, 2015-02-20:
Couldn’t the system try to estimate the battery resistance and voltage drop on its own? If you read the voltage when it starts, record it and wait for the load to ramp (such as when the motors start), you should be able to get the difference (the drop). From there you can (guess)estimate the battery internal resistance (v0 - v1) / (a1 - a0). With that information, we would be able to better tell the battery condition under load and nominal no-load voltage (with the data stored in `battery_status_s`  and sent across with the rest of battery data). Or maybe I’m just thinking too much and too tired to think straight… :)

LorenzMeier, 2015-02-20:
In theory yes - similar to that double-integrating acceleration should yield position - but it doesn't hold up for more than a few seconds, because of offsets and rotation errors.
You would have to try how accurate you can get. Meanwhile we're exploring smart batteries via SMBus, which seems to work great.

LorenzMeier, 2015-06-14:
Closing in favor of smart batteries.

closed 2015-06-14 by LorenzMeier