PX4/PX4-Autopilot #1843: RAM Cleanup

LorenzMeier, 2015-02-27:
* mc_pos_control_multiplatform: Reduction from 3K to 1.8K?
  * mc_att_control_multiplatform: Reduction from 3K to 1.8K?

LorenzMeier, 2015-02-27:
For the non-multiplatform apps #1844 implements a reasonable cleanup. @sjwilks Testing would be great, in particular since it gets you back to a normal operation mode on FMUv1.

SimonWilks, 2015-02-28:
I'll flash this branch now and give it a test flight.

SimonWilks, 2015-02-28:
Here are the results after arming with two mavlink instances running:

```
Processes: 21 total, 3 running, 18 sleeping
CPU usage: 33.54% tasks, 0.31% sched, 66.15% idle
Uptime: 211.443s total, 138.332s idle
   2 lpwork                        650  0.291   364/ 1792  50 ( 50)  w:sig 
 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                  138331 66.149     0/    0   0 (  0)  READY 
   1 hpwork                       4716  2.202   788/ 1792 192 (192)  w:sig 
   2 lpwork                        664  0.310   364/ 1792  50 ( 50)  READY 
   3 init                          315  0.000  1428/ 2592 100 (100)  w:sem 
   6 nshterm                        15  0.000   868/ 1496  70 ( 70)  w:sem 
 153 top                          1196  0.310  1204/ 1696 100 (100)  RUN   
  63 dataman                       321  0.000   612/ 1792 100 (100)  w:sem 
  69 commander                     480  0.186  2740/ 3192 215 (215)  w:sig 
  71 commander_low_prio             36  0.000   476/ 2592  50 ( 50)  w:sem 
  73 px4io                        3066  1.458   776/ 1792 240 (240)  w:sem 
  78 mavlink_if0                  1574  0.682  1780/ 2696 100 (100)  READY 
  79 mavlink_rcv_if0                46  0.000   780/ 2096 175 (175)  w:sem 
  82 mavlink_if1                  7572  3.568  1716/ 2696 100 (100)  READY 
  83 mavlink_rcv_if1               285  0.186   780/ 2096 175 (175)  w:sem 
 107 sensors_task                 4919  2.327  1284/ 1992 250 (250)  w:sem 
 125 ekf_att_pos_estimator       41417 19.826  3708/ 7496 215 (215)  w:sem 
 117 gps                           365  0.155   820/ 1496 220 (220)  w:sig 
 132 mc_att_control               1871  0.868   996/ 1792 250 (250)  w:sem 
 138 mc_pos_control               1389  0.651  1116/ 1792 250 (250)  w:sem 
 141 land_detector                 388  0.155   548/  992 100 (100)  READY 
 148 navigator                    1540  0.651   756/ 1792 120 (120)  w:sem 
nsh> free
             total       used       free    largest
Mem:        173760     164016       9744       5648
```

Heading out for some real flying now.

SimonWilks, 2015-02-28:
With 2 mavlink instances running free memory goes down to 9.1Kb after arming, that's the lowest I have seen. I couldn't test RTL and auto as I couldn't get a GPS fix (near trees) but I did back at home and flew in POSCTL on the balcony so I assuming more memory won't be consumed when in auto?

What I also had difficulty with was connecting with QGC over USB and couldn't connect though that didn't change with or without the second mavlink running so this may be something else or just a local issue. Over the radio was fine though.

Conclusion: free RAM is pretty tight with 2 x mavlink instances running (no surprise) but everything related to this change seems to work.

LorenzMeier, 2015-02-28:
Could you also run the off board control stuff (or did it run already?)? I'm concerned that this could let the MAVLink stack usage skyrocket.

LorenzMeier, 2015-02-28:
And sorry for not being more friendly, your testing is much appreciated!

SimonWilks, 2015-02-28:
I launched the mavros node that communicates withe the PX4 though I didn't
run the offboard demo app. Can do that tomorrow.

SimonWilks, 2015-03-01:
This time I ran the offboard demo app on the companion computer and there weren't any issues.

thomasgubler, 2015-03-03:
fixed in #1858

closed 2015-03-03 by thomasgubler