PX4/PX4-Autopilot #1851: px4io: check CRC failed: -22: Invalid argument

velocoderaptor, 2015-02-28:
Branch: stable
Commit: 6d1df8f

From NuttX shell during boot:

	nsh> [boot] Initialized SPI port 1 (SENSORS)
	[boot] Initialized SPI port 2 (RAMTRON FRAM)
	[boot] Initialized SPI port 4
	[boot] Initialized SDIO
	sercon: Registering CDC/ACM serial driver
	sercon: Successfully registered the CDC/ACM serial driver
	[i] Looking for microSD... 
	[i] microSD mounted: /fs/microsd 
	[i] Init script not found: /fs/microsd/etc/rc.txt 
	nshterm [6:70]
	param: selected parameter default file /fs/mtd_params
	[param] Loaded: /fs/mtd_params 
	[rgbled] on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
	nsh: pca8574: command not found
	 1:   SYS_USE_IO: match
	 4011: + SYS_AUTOSTART: match
	[i] Config not found: /fs/microsd/etc/config.txt 
	px4io: check CRC failed: -22: Invalid argument
	telem> config for RC6 rejected by IO
	[px4io] default PWM output device
	px4io: failed to enable safety

velocoderaptor, 2015-02-28:
I think the problem is the `px4io safety_on` command.
Since it's failing, the startup script immediately exits and never gets to the part where it does a `forceupdate`.

velocoderaptor, 2015-02-28:
Ah, was able to reproduce it. I saved my previous parameters before I flashed to stable.
Loaded back 3454c42, restored all the params, and then loaded stable again.
And now I see the error again.

velocoderaptor, 2015-02-28:
Looking back, this was introduced with commit eb28aa6

LorenzMeier, 2015-02-28:
Sorry about that. I've pushed 6646e3b to fix this. I'm closing this, please let me know in case the issue persists.

zottgrammes, 2015-03-02:
Thanks. That put me past the `px4io safety_on` failure, but the bulid is still not flashing on my Pixhawk.

The `px4io_uploader.cpp` code fails to get a sync from the bootloader:

	px4io: check CRC failed: -22: Invalid argument
	[px4io] default PWM output device
	px4io: failed to enable safety
	[PX4IO] using firmware from /etc/extras/px4io-v2_default.bin
	[PX4IO] looking for bootloader
	[PX4IO] bad sync 0xff,0xff
	[PX4IO] bad sync 0xff,0xff
	[PX4IO] bad sync 0xff,0xff
	[PX4IO] bad sync 0xff,0xff
	[PX4IO] bad sync 0xff,0xff
	[PX4IO] bad sync 0xff,0xff
	[PX4IO] bad sync 0xff,0xff
	[PX4IO] bad sync 0xff,0xff
	[PX4IO] bad sync 0xff,0xff
	[PX4IO] bad sync 0xff,0xff
	[PX4IO] bad sync 0xff,0xff
	[PX4IO] bad sync 0xff,0xff
	[PX4IO] bad sync 0xff,0xff
	[PX4IO] bad sync 0xff,0xff
	[PX4IO] bad sync 0xff,0xff
	[PX4IO] bootloader not responding
	px4io: check CRC failed: -22: Invalid argument
	[i] ERROR: PX4IO not found
	[i] ERROR: PX4IO not found, disabling output

Fails here: https://github.com/PX4/Firmware/blob/master/src/drivers/px4io/px4io_uploader.cpp#L138

LorenzMeier, 2015-03-02:
Does current master still fail for you?

zottgrammes, 2015-03-02:
I cherry picked 6646e3b onto my local copy of stable.

I'll flash master and try that.

zottgrammes, 2015-03-02:
Yeah, same problem on master: 8424325

LorenzMeier, 2015-03-02:
Bugger - I have not yet managed to reproduce. Could you look into this and maybe provide a patch which works locally?

zottgrammes, 2015-03-04:
I think I found it. px4io needs to be stopped when `px4io safety_on` fails. Not clear on the reason, I suspect there's a conflict with trying to access the bootloader and px4io itself.

Code snippet:

	if px4io start
	then
		# try to safe px4 io so motor outputs dont go crazy
		if px4io safety_on
		then
			# success! no-op
		else
			# px4io couldnt be safed, just shut it down
			px4io stop
		fi
	fi

LorenzMeier, 2015-03-04:
Thanks! The reason I missed this is that this only occurs if IO doesn't understand the safety command yet. So this happens mostly for people which update from a state a few weeks or months old, and it doesn't happen for people updating regularly. Sorry about that.
https://github.com/PX4/Firmware/pull/1879 fixes it.

LorenzMeier, 2015-03-08:
Could you please try master and report back?

zottgrammes, 2015-03-09:
I cherry picked this on my stable and it works. Thanks!
Do you stil lneed verificaiton on master branch?

closed 2015-04-03 by LorenzMeier