PX4/PX4-Autopilot #1856: Offboard velocity setpoints broken on master

thomasgubler, 2015-03-01:
We have reports that https://github.com/PX4/Firmware/pull/1786 introduced a regression that disables offboard velocity setpoints.

More tests are needed.

blackcoder, 2015-03-11:
@thomasgubler did you find a solution for this bug ?

thomasgubler, 2015-03-11:
No, I didn't see it directly in the code and I currently don't have the setup ant time to reproduce it.

blackcoder, 2015-03-12:
No worries, I had a quick look at the code, nothing is obvious there, the logic is correct, it must be some hard to find bug.

xuliren, 2015-05-18:
Hi, I'm testing the offboard velocity function under an indoor-vicon testbed. Currently the uav can do the offboard position control very well, but the velocity control are not working. After reading the source code, I found that https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L715 is converting the velocity control into an position control, I think it might not that proper to to so.

supermice, 2015-06-28:
@xuliren  What's the status now? Does the master firmware fixed the velocity control problem?

aerialhedgehog, 2015-07-01:
I tried flying this yesterday and did not get any response from my hex
while sending velocity commands.

aerialhedgehog, 2015-07-01:
i was flying master, using the c_uart_interface example
https://github.com/mavlink/c_uart_interface_example

AndreasAntener, 2015-07-03:
I think I found the missing piece: #2527

PR #2527 (2015-07-04): Set altitude control flag for velocity control
@LorenzMeier @thomasgubler The only thing missing for velocity control was the right control flag setting, fixes #1856

Tested in SITL. Can anyone test this on a real one?
closed 2015-07-03 by LorenzMeier