PX4/PX4-Autopilot #1861: Better status feedback via MAVLink

LorenzMeier, 2015-03-02:
* Populate status flags in VEHICLE_STATUS
  * Send AUTOPILOT_VERSION on boot
  * Send AUTOPILOT_VERSION on MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES command.

DonLakeFlyer, 2015-03-02:
Just tracking this one since I can add autopilot version to vehicle summary in QGC when it shows up.

tumbili, 2015-05-15:
@LorenzMeier From what I see the status flag population is already done, it's just that not every sensor publishes the "subsystem_info" topic yet.

LorenzMeier, 2015-05-15:
Yeah, that subsystem_info topic might not be the ultimate solution. I have to think about that one.

TSC21, 2015-05-22:
mavros is waiting for this message support also: https://github.com/mavlink/mavros/issues/246

LorenzMeier, 2015-05-23:
I'm closing this for now since the current reporting is already decent. We can still improve long term.

closed 2015-05-23 by LorenzMeier