PX4/PX4-Autopilot #1887: coordinate frame for vehicle_local_position and vehicle_local_position_setpoint

thedinuka, 2015-03-06:
I've been trying to get mc_pos_contrl to work with my vision based position estimation solution. 
For that to work, I need to know the coordinate frame in which vehicle_local_position should be published in.

More specifically, I'm assuming vehicle_local_position is the position of the vehicle origin w.r.t. some earth fixed coordinate frame. However, this position can be **expressed** in either the earth fixed coordinate frame or the body corrdinate frame of the vehicle. Due to the "local" qualifier in vehicle_local_position, I'm assuming it is the latter. Is that right?

Second, what is the standard used for the orientation of the earth fixed coordinate frame (Maybe this is called the global coordinate frame?) Is it with Z axis pointing down troward the earth or up?

LorenzMeier, 2015-03-08:
The coordinate frame is in the whole system NED.

thedinuka, 2015-03-09:
@LorenzMeier, thanks for the input. While that is a useful bit of information, my question was more about the coordinate frame in which the vehicle_local_position should be expressed. To be more specific, it can be expressed in the earth fixed NED coordinate frame or the body fixed Front-Right-Down coordinate frame. These two can be considerably different when the body frame does not alight with earth fixed frame.

kd0aij, 2015-03-10:
I think you'll find that vehicle_local_position is expressed in the earth fixed NED coordinate frame  in units of X,Y,Z meters relative to the "home" position. Whereas the global position  is expressed using latitude, longitude and MSL altitude.

closed 2015-06-14 by LorenzMeier