PX4/PX4-Autopilot #1898: No PWM in standby though param is set

friekopter, 2015-03-09:
Though configuration introduced in  #1446 is still in place, pwm signal in disarmed mode is not sent any more.

# Provide ESC a constant 1000 us pulse while disarmed
set PWM_OUT 4
set PWM_DISARMED 1000

friekopter, 2015-03-13:
Looks like 653b3e71be7059707a0963e15dfb12e5403025aa breaks this.

LorenzMeier, 2015-03-15:
Thanks for narrowing this down. But there are more commits in that series. Could you have a look at pwm.c and see if you can resolve it for your case? The diff shows you which lines to look at, and it should be fairly easy with a few printf / warnx() statements.

friekopter, 2015-03-15:
Try to do that, but currently I am not sure any more if this change produced the effect I have. Reverting that change does not always fix my problem, it is somehow undeterministic :-(

Date: Sun, 15 Mar 2015 10:14:56 -0700

tumbili, 2015-04-09:
@LorenzMeier Is there some kind of timeout for the disarmed pwm value? When I reboot the Pixhawk I can measure the disarmed pwm pulse but after a minute or so I can't measure it anymore. Then if I do "pwm test -c 1 -p 1500" I can measure the disarmed pwm value again.

LorenzMeier, 2015-04-09:
There shouldn't be a timeout, if there is one its something we have to investigate.

tumbili, 2015-04-09:
@LorenzMeier Sorry, I had "armed" mistaken with "safety off". If safety is off I don't get the timeout.

friekopter, 2015-04-12:
In my case the problem is gone with newest checkout from master.

tumbili, 2015-04-19:
One issue was found to be related to the attitude_controller_ekf module. At some point it stopped sending attitude messages which lead to the PX4IO claiming FMU_FAIL. In my case the attitude estimator was provided with a zero mag vector which eventually lead to a NaN in the attitude estimate. I think the problem with the zero mag vector was fixed in (3658bf8). I could not reproduce the problem with ekf_att_pos_estimator.

tumbili, 2015-04-29:
I think this can be closed.

closed 2015-05-13 by tumbili