PX4/PX4-Autopilot #1899: HMC5883 driver start failure on i2c

kalnir, 2015-03-09:
An external magnetometer (HMC5883L) is connected via the i2c bus to the pixhawk. On using nuttshell to start the driver the following happens:

nsh> hmc5883 start
hmc5883: no device on bus 1
[HMC5883_SPI] on SPI bus 1 at 5 (11000 KHz)
hmc5883: no device on bus 3
hmc5883: driver start failed

Same issue repeated for two different external GPS+Compass modules using the HMC5883L.

LorenzMeier, 2015-03-27:
Could you repeat your test against master? I wouldn't know why it should not work.

tumbili, 2015-03-28:
I think the problem is that if you just call "hmc5883 start" then the driver will try to start every bus (internal & external I2C & SPI). Bus 1 is the internal I2C bus which probably has nothing connected.
Try hmc5883 -X start to start explicitly the external I2C bus. For me this works.

kalnir, 2015-03-30:
Using the -X flag works and the external mag works on the latest master.

closed 2015-03-30 by kalnir