PX4/PX4-Autopilot #1902: Evaluate att_post_estimator_ekf performance

LorenzMeier, 2015-03-10:
@Zefz and I noticed that the overall performance degraded with the latest bugfixes. The current state needs to be carefully analyzed again on the test bench.

Another log to look at is this one:
http://dash.oznet.ch/view/u4eKwNr8ziKVDUXmwJo933#Thrust_PLOT

Zefz, 2015-03-10:
Quoting myself from chat:

Perhaps the issue is in AttPosEKF::ResetPosition() which does not reset covariance values for position
Previously when the filter accepted reset position estimate after rejection, it reset a lot of the filter and cleared much data (causing the sudden jump glitch on GPS regain among other things). Perhaps some of the data clearing/reset (like covariance) need to be added back in

Zefz, 2015-03-19:
#1913 and #1919 seem to have resolved the major EKF issues we were experiencing. Is the filtering performance still degraded?

LorenzMeier, 2015-03-19:
Yes, we're almost there. I would love to have another look at the Z estimation performance, since the filter output should be better than what the baro provides.

Zefz, 2015-04-17:
According to this post in the mailing list the Z estimation still has some issues:
https://groups.google.com/forum/#!topic/px4users/Jur7I82lVcg

Here is some log data using EKF:
http://dash.oznet.ch/view/QHbHZsgPoixbWhcbxYVUoj

Notice that altitude estimations has clearly some oscillations that are induced by the filter itself.

Compare this with flying the same quad using the INAV filter:
http://dash.oznet.ch/view/DF8S3URq88sB8SfHucpZ49#Local_Z_PLOT

LorenzMeier, 2016-03-24:
Doing that, moving to EKF2

closed 2016-03-24 by LorenzMeier