PX4/PX4-Autopilot #1910: Script/Code Duplication in ROS/SITL install scripts

thomasgubler, 2015-03-11:
@AndreasAntener We need to find a way to have only one script for dependencies etc.

https://github.com/PX4/containers/tree/master/docker/ros-indigo/px4-ros-full

vs.

https://github.com/PX4/Firmware/tree/master/Tools/ros

AndreasAntener, 2015-03-11:
Yeah that's a pity..
Maybe we can sync the scripts and I can reference them from the Docker installation. But I have to check how, because Docker only allows using stuff from sub directories (where the Dockerfile is located).

AndreasAntener, 2015-04-04:
@thomasgubler I'll try move the install scripts into PX4/containers and reference it from there. So I can use them for the Docker installation. At the same time I'll try to switch to latest MAVROS because of #1979

closed 2015-04-08 by thomasgubler