PX4/PX4-Autopilot #1915: Random mavros_attctl errors on SITL CI

AndreasAntener, 2015-03-13:
This seems to be related to a timing issue with MAVROS and publishing attitude and throttle. Sometimes I see the same timestamps for alternating attitude_setpoints (att/thrust).

Will try to fix it with the next SITL demo test update

AndreasAntener, 2015-03-15:
To verify my timing suspicion I added some debug output in the attitude controller ([here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_att_control_multiplatform/mc_att_control_base.cpp#L86)):
```c
warnx("thrust: %f", _v_att_sp->data().thrust);
warnx("att sp: %f, %f, %f", _v_att_sp->data().roll_body, _v_att_sp->data().pitch_body, _v_att_sp->data().yaw_body);
```

I then executed the test a few times like it is at the moment, and then a few times with a small sleep time between publishing attitude and thrust.

**Without sleep:** 
https://gist.github.com/AndreasAntener/6378d324e9e7d72c819d#file-failed1-txt
It shows that for a large amount of time, the attitude is missing, and the copter just climbs vertically (almost only). The test fails more or less every 3rd time.

**With sleep:**
https://gist.github.com/AndreasAntener/6378d324e9e7d72c819d#file-success1-txt
The values are much more mixed now, and it does what I expect it to do. The test always succeeds.

thomasgubler, 2015-03-15:
From gitter chat:
>AndreasAntener 4 minutes ago  
All the vars there get freshly initialized
thomasgubler 4 minutes ago  
yes thats what I mean, https://github.com/PX4/Firmware/blob/master/src/platforms/ros/nodes/mavlink/mavlink.cpp#L174 and aboce should only happen if attitude was really updated
so instead of if (!offboard_control_mode.ignore_attitude) { just if (!ignore_attitude) {
and add the same check to https://github.com/PX4/Firmware/blob/master/src/platforms/ros/nodes/mavlink/mavlink.cpp#L165 and https://github.com/PX4/Firmware/blob/master/src/platforms/ros/nodes/mavlink/mavlink.cpp#L167

The same applies to https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp

LorenzMeier, 2015-03-15:
@thomasgubler Could you send a quick PR for the mavlink receiver?

AndreasAntener, 2015-03-15:
oh, I only updated the multiplatform node, I'll leave the real one to the experts ;)

AndreasAntener, 2015-03-15:
> @AndreasAntener I updated your initial PR description:

ok :)

Added the members

PR #1920 (2015-03-15): Fixing handling of attitude setpoints
Fixes #1915
closed 2015-03-15 by thomasgubler