PX4/PX4-Autopilot #1916: Make EKF code standalone again

LorenzMeier, 2015-03-14:
@Zefz This change here makes the EKF code non-standalone because it links it to the GEO lib. This is something we want to avoid in general, because much of the robustness / quality of the filter stem from the testing in the test harness:
https://github.com/Zefz/Firmware/commit/67695f191ea6f528971c6d2da3dcac66540d7be9

We will either need to undo the change or move the GEO library into the filter.

I will refactor the filter repo soon and then include it as submodule. This will allow us to use it in a single repo and ensure the test-harness tests are run on changes, which will in the future also greatly help with stability.

Zefz, 2015-03-14:
I see. However, geo lib projection was already used here:
https://github.com/Zefz/Firmware/blob/67695f191ea6f528971c6d2da3dcac66540d7be9/src/modules/ekf_att_pos_estimator/ekf_att_pos_estimator_main.cpp#L845

and I thought it would make sense that the same projection logic was used for input as well as output to minimize any discrepancy between projection calculations. So either we should use only geo for projection or not use geo projection at all? (and have the EKF implement its own projection as before)

LorenzMeier, 2015-04-03:
Done in master. Closing.

closed 2015-04-03 by LorenzMeier