PX4/PX4-Autopilot #1925: Mixer: Re-evaluate limit enforcement

LorenzMeier, 2015-03-16:
Reported by @kd0aij. At thrust close to 1 the actuator outputs seem to dip to min.

[image]

kd0aij, 2015-03-18:
Unable to reproduce the problem in manual attitude control mode:
[image]
commit 59ebc99,  branch limitAutoRPrates

LorenzMeier, 2015-03-19:
I'm not surprised that you can't, because for the saturation to occur you would need to set rates instantaneously, which is is almost impossible by just twiddling an RC stick.

kd0aij, 2015-03-19:
Just rechecked the logs: same mode in both, manual attitude control

tumbili, 2015-03-19:
@kd0aij Can you share a link to the log.

kd0aij, 2015-03-19:
first log: https://copy.com/dZ6jHciEUuhOULZo
second log: https://copy.com/ekAYIlWe29NdGPpN

kd0aij, 2015-03-19:
BTW, I've accidentally caused integer overflows in MatrixPilot, which is mostly implemented on DSP hardware; results are usually quite exciting for a very short time :)

LorenzMeier, 2015-04-03:
@tumbili The proposed approach is along the lines of this algorithm. You need to implement it partially at least in the controller to lock integrals from building up.

CASE A: Slew rate limiting: Last attitude setpoint and current attitude setpoint differ by more than e.g. 30 degrees OR the requested angular rate is relatively high AND the mixer reports hitting a limit:
  * Limit demanded angular rate to a fixed threshold in controller (e.g. 200 dps)
  * Limit demanded attitude to an attitude max. 15 deg or so from current attitude
  * Lock all integral gains

This forces the system into a slew rate limiting mode and ensures that higher-level controllers do not build up integrals.

CASE B: Flip-prevention: Current attitude and setpoint differ by e.g. more than 20 degrees (NOTE: The limit in degrees here has to be higher than in slew rate limit mode so that slew rate limiting always triggers first). This mode might loose or gain altitude.

  * If hitting high limit: Reduce throttle to 60-80% of the demanded value to allow higher spread
  * If hitting low limit: Increase throttle to 120-160% of the current value

This should allow the system to recover if it was close to flipping.

LorenzMeier, 2015-05-09:
Fixed in master.

closed 2015-05-09 by LorenzMeier