PX4/PX4-Autopilot #1945: fixed wing - servo oszillation / jittering only in manual-mode!

severinleuenberger, 2015-03-23:
I have some serious servo jittering - not only small oszillations on manual mode. 
The strange thing is, that I do not have any interferences around center position.
And I do also not have any jittering/interferences on ALTCTL.

I tried to connect the Servos directly to the Transmitter (x8r), and there is no interference.

Interference are there on manual mode only. 
Testing on changing antenna positions and cables do not chnage anything. 

I use digital servos on a pwm signal. And I tried with changing the hz-signal-rate (tested on 50, 200, 

330 and 400hz). Doesnt help anything. 

Then I changed the PX4 firmware from the actual dev master build to the stable build. Same problem, 

nothing changed. 

Therefore I think, it is a problem from PX4 Firmware, that causes these interferences. 

Can anyone look at that problem?

Best whishes
Severin

severinleuenberger, 2015-03-23:
this ist the *.mix I used: 

M: 2
O:      10000  10000      0 -10000  10000
S: 0 0   7500   2500      0 -10000  10000
S: 0 1  -4000  -4000      0 -10000  10000

M: 2
O:      10000  10000      0 -10000  10000
S: 0 0   4000   4000      0 -10000  10000
S: 0 1  -8000  -8000      0 -10000  10000

M: 2
O:      10000  10000      0 -10000  10000
S: 0 0  -4000  -4000      0 -10000  10000
S: 0 1  -8000  -8000      0 -10000  10000

M: 2
O:      10000  10000      0 -10000  10000
S: 0 0   2500   7500      0 -10000  10000
S: 0 1   4000   4000      0 -10000  10000

M: 1
O:      10000  10000      0 -10000  10000
S: 0 2  10000  10000      0 -10000  10000

M: 1
O:      10000  10000      0 -10000  10000
S: 0 2  10000  10000      0 -10000  10000

M: 1
O:      10000  10000      0 -10000  10000
S: 0 3      0  20000 -10000 -10000  10000

M: 1
O:      10000  10000      0 -10000  10000
S: 0 4  10000  10000      0 -10000  10000

bansiesta, 2015-03-23:
If you have a logfile please share it because it should show where the jitter appears and where it does not.

severinleuenberger, 2015-03-24:
I uploaded some logs on my onedrive here: 

https://onedrive.live.com/redir?resid=6A7B30E7A53DBC6D!55283&authkey=!AIL4mDZTx1qw3Is&ithint=folder%2cpx4log

severinleuenberger, 2015-03-24:
I just performed the following tests: 

- connected  the 8 analog servos to 2 "sbus to pwm converter" and connect the converter to pixhawk sbus out. Same result, incredible jittering on the servos (but not in center and not in altctl). 
- to be sure that there is nothing with the servos, I connected the servos via SBUS to x8r. Everything runs smooth. 
So I am pretty sure, that there is a BUG in either px4 firmware or the Pixhawk Dropix v1.2 from drotek. 

Can you help me, please?
Severin

bansiesta, 2015-03-25:
Thanks for the logs, that's helpful.

Until now I can only confirm your issue: It looks like the RC input is ok but there is jitter on the output in manual mode.

[image]
[image]

severinleuenberger, 2015-03-25:
I just tested with an other dropixv1.2 (pixhawk) and there i do not have the jitter!!!
So then it could be a setting in px4 who does the jitter or a hardware failure. 
How can I compare with a simple method where the differences are in the settings?

severinleuenberger, 2015-03-25:
Here you can see the differences between the 2 files. I do not see an obvious reason...
https://www.diffchecker.com/v5wso150

severinleuenberger, 2015-03-27:
I talked with Jerome Perin from drotek france. I will send the other dropix back so they can check, if it is a hardware problem, that causes the jitter. I will inform here again, what the findings are from drotek.

severinleuenberger, 2015-04-20:
My newest finding here is: If you flash the stable, build: no jittering. 
If you then flash to the newest dev-build: no jittering
If you go back and flash the stable build - THEN YOU HAVE THE SERVO JITTERING
If you go to the newest dev-build - then the jittering is gone again. 

Strange, but because I do not go back to stable release, there is no issue for me anymore. 
(When will be the next officially announced new stable release? )

bansiesta, 2015-04-26:
Does this also happen if you make sure the IO is updated? If you press the arming switch when you connect power, it will force an IO update.

severinleuenberger, 2015-04-26:
Hello baniesta, thanks for the hint with forcing IO update.

because I am on the newest px4 dev build, and do not plan to go back on stable release, i would say, that we can close this issue and I will report, if something similar happens with the new px4 dev build.

closed 2015-04-26 by severinleuenberger