PX4/PX4-Autopilot #1967: windows toolchain installer nsis error

joeyfezster, 2015-04-01:
Hey, i've tried downloading the windows toolchain installer from 3 different computers, and still get the same error.

is there a different source i could get it from?

joeyfezster, 2015-04-01:
FIX: control panel -> region -> administrative -> change system locale -> choose acording to local language -> restart pc

LorenzMeier, 2015-04-03:
Thanks for the solution - but you haven't shared the error in the first place?

joeyfezster, 2015-04-03:
i just opened the installer, and it gave the nsis error in a message box.
no details.

the fix is based on previous encounters with the same error

closed 2015-07-03 by LorenzMeier

alielectric, 2017-06-07:
your location that the installer is located has more than a defined characters, copy it some where else than run the setup