PX4/PX4-Autopilot #1970: The px4flow board ,debugging ,  Didn‘t boot to jump to app ?

molyswu, 2015-04-02:
LorenzMeier !
I used STLINK-V2(JTAG) full chip erase px4flow board and build px4 Flow-master file (2015-01-24 Flow-master)
, make all Flow(firmware) for px4flow_bl.elf and px4flow.bin , compiler code Successful and connect STLINK-V2 JTAG ,then STLINK-V2 program px4flow_bl.elf or px4flow.bin to px4flow board in Flow(c/px4/Flow/).restart the board show "USB device not recognized" . the pxflow board idn't communicated with PC USB, The three LED is lit.Is the problem Bootloader of code jump to app? I was under the windows compiler and debug PX4 Toolchain v14/eclipse/clonse ,try to used tool for Toolchain Eclipe (GDB SEGGER J-LINK debugging) ,Debug results are same show "USB device not recognized" )  
as a attachment GDB SEGGER J-LINK debugging:
[image]

I was checked STM32F407VG "clock_setup " with GPIO 12PIN ,13PIN ，voltage only have 0.7V. Righten the tow PIN voltage have 1.7V。boot didn‘t jump to app.Ho to process the problem?
Thank you very much for your help!

cctsao1008, 2015-04-02:
would please try my solution from following thread
https://groups.google.com/forum/m/#!topic/px4users/B4-wLZoc5XY

davids5, 2015-04-06:
Hi @molyswu,

There was a change made in https://github.com/PX4/Bootloader/commit/b75a627a23e97596566faf31044bff1a494b0ec6 

Are you using that version or later? If not, please use git pull and see if your problem has been resolved. 

I noticed you posted 
```
would please try my solution from following thread
https://groups.google.com/forum/m/#!topic/px4users/B4-wLZoc5XY
```

But it was removed. Is this because the issue was **not** fixed by the code changes you described in the link?

Please post back your results.

LorenzMeier, 2015-06-25:
Closing as we have no feedback.

closed 2015-06-25 by LorenzMeier