PX4/PX4-Autopilot #1978: ROS SITL tutorial with Python script

LorenzMeier, 2015-04-03:
@thomasgubler @AndreasAntener  it would be nice to pull out one of the auto-test scripts and create a launch file + Wiki docs for it. This should be a quick fix for one of you, and would lower the barrier greatly. The joystick setup turned out to be very fiddly and its a source of great frustration on the initial SITL setup as I found out during my week in the Bay. Nobody was able to get it up and running.

thomasgubler, 2015-04-03:
So basically a setup which does not require the joystick such as the offboard position control demo?

```
roslaunch px4 gazebo_ardrone_empty_world_offboard_positiondemo.launch 
```

AndreasAntener, 2015-04-03:
Hey, just yesterday I thought the same about a starting point for people who want to do offboard control with mavros.
@thomasgubler yeah, something like the gazebo_ardrone_empty_world_offboard_positiondemo.launch but without the demo node (and without joystick). And possible one or two scripts or CPP nodes that you can then launch separately to show it doing something. I also have a script that let's you fly it in POSCTL by keyboard that we could add too.

LorenzMeier, 2015-04-04:
Yes, essentially just a simple launch file that "does something" would be great. The POSCTL flight with keyboard sounds also cool. Essentially anything that allows a fool-proof first SITL experience.

thomasgubler, 2015-04-04:
@AndreasAntener why without the demo node? I agree that the demo node is not very useful but it would give that "fool-proof" experience. Of course I am not  speaking against a more advanced demo with keyboard control etc.

AndreasAntener, 2015-04-04:
The problem with the demo node arises as soon as someone has it running and then decides start a diffrent script/node that does something (his own or from us). Then you need to explain how to kill the already running demo node or modify the launch file (sounds simple I know, but I witnessed people doing far more with SITL before they started to learn ROS basics).. We could have 2 launch files, one like the posctl demo we already have and one without.

thomasgubler, 2015-04-04:
ok or we create 2 launch files: one to start SITL (quad just sits there
waiting) and a second one which only launches the demo node. The wiki would
instruct the user to start both. This way it should be clear that it's the
demo node which does the "intelligence" and the users can start to fiddle
around with it.

LorenzMeier, 2015-04-04:
Makes sense to me.

AndreasAntener, 2015-04-04:
Sounds good.
I can still provide the keyboard script as a demo option.

AndreasAntener, 2015-04-05:
Oh, to launch the demo node we wouldn't even require another launch file, could just instruct to use `rosrun`

LorenzMeier, 2016-02-18:
We have this more or less.

closed 2016-02-18 by LorenzMeier