PX4/PX4-Autopilot #1979: Test and fix SITL with latest MAVROS

AndreasAntener, 2015-04-03:
Lots of changes in latest MAVROS, e.g. changed plugin configuration and renamed topics
Need to update launch files and tests.

vooon, 2015-04-07:
Hmm, update rate for ros repo is too slow... Shadow repo faster but "unstable".

New things to use in python scripts in 0.11.1:

1. one global place for handling namespace:
```python
import mavros

mavros.set_namespace('/iris/mavros') # by default it is '/mavros'
print mavros.get_topic('gps', 'fix') # -> /iris/mavros/gps/fix
```

2. some plugins now have python module:
```
import mavros
from mavros import setpoint as sp

mavros.set_namespace('/iris/mavros')
pub = sp.get_pub_position_local(queue_size=10, latch=False)
```

AndreasAntener, 2015-04-07:
Thanks for the hints!
- How are the shadow repositories "unstable"? (what do I have to expect?)
- What's the advantage of using  `set_namespace` over just putting MAVROS inside a namespace with the a launch file group tag (like we do now)?
- Regarding the plugin python modules: I'm not sure if it's a good idea to hide the ROS API inside the plugins. It should rather just provide the namespace and message type, but the user should use ROS API directly for creating the publisher.

vooon, 2015-04-07:
Shadow not recommended in ros wiki, usually i only update mavlink and mavros from that place (only travis build uses all packages from shadow).

Mavros can be started with different node namespace, e.g. see px4flow.launch and with `set_namespace()` you only need change ns, instead of all this `iris/mavros/xyz`.

Maybe, but this reduce copy-paste count.

AndreasAntener, 2015-04-07:
> Mavros can be started with different node namespace, e.g. see px4flow.launch and with set_namespace() you only need change ns, instead of all this iris/mavros/xyz.
I get the same effect with the group tag and ns parameter. This only copies basic launch file functionality. And, although I'm not sure about this, but it seems that if you would have multiple nodes that use `set_namespace()` they'll overwrite each other.

vooon, 2015-04-07:
Yes, `set_namespace()` is global, so you cant have multiple in one node (python process).
But for most possible use-cases we need only one mavros-script connection.

How `<group>` may change ns if we run `<node type="mavros_node" name="some_different_ns" ... />`
(e.g. paths for topics: `some_different_ns/state` instead of `mavros/state`).

AndreasAntener, 2015-04-07:
If you look at [gazebo_ardrone_empty_world_offboard_positiondemo.launch](https://github.com/PX4/Firmware/blob/master/launch/gazebo_ardrone_empty_world_offboard_positiondemo.launch) you see there that we declare `ardrone` in the top-level launch file. The included [mavros_sitl.launch](https://github.com/PX4/Firmware/blob/master/launch/mavros_sitl.launch) uses this argument to put the mavros nodes inside this namespace. In [code](https://github.com/PX4/Firmware/blob/master/src/platforms/ros/nodes/demo_offboard_position_setpoints/demo_offboard_position_setpoints.cpp#L49) we then write `mavros/setpoint_position/local`. Of course I understand that I cannot change the `mavros` part like this, but why would I want to ;)

PR #1988 (2015-04-08): Mavros update
**WIP**
@thomasgubler : this updates tests and demo nodes for MAVROS 0.11
More fixes are coming (another NS change) as soon as I get 0.11.1 on Ubuntu, will wait for that before merging.
Fixes #1979
closed 2015-04-08 by AndreasAntener